tpaviot/pythonocc-core #443: About STLAPI

qqnzhyxxx, 2017-07-18:
I got a little problem that when I use this code try to transform a stp file to stl.
After running ,I find the **state = 1**,it means StlAPI_StatusOK or not?
But I cant find the output file **test.stl**. I dont what cause this problem.
Thanks for your reading,and waiting for your relpy.
`from OCC.STEPControl import STEPControl_Reader
from OCC.StlAPI import StlAPI_Writer

step_read = STEPControl_Reader()
step_read.ReadFile('lft.stp')
step_read.TransferRoot()
stp_shap = step_read.Shape()
stl_write = StlAPI_Writer()
stl_write.SetASCIIMode(True)
state = stl_write.Write(stp_shap, 'test.stl')
print(state )`

qqnzhyxxx, 2017-07-19:
I have find answer in your examples.
in the function **StlAPI_Writer.Write(stp_shap, 'test.stl')**, the fist praram is not a simple Shape,
But a shap had been meshed.
I add codes before Write(), then  it works. 
Thanks for your examples. 
`from OCC.BRepMesh import BRepMesh_IncrementalMesh
mesh = BRepMesh_IncrementalMesh(stp_shap, 0.9)
mesh.Perform()
`

tpaviot, 2017-07-25:
@qqnzhyxxx absolutely, the StlAPI_Writer class now takes a meshed shape, this is something that changed in oce-0.18. This way, you can easily tune the mesh before it is serialized to an stl file.

closed 2017-07-25 by tpaviot