tpaviot/pythonocc-core #444: ImportError: No module named OCC.TColgp

brunozilli, 2017-07-27:
Hello Thomas,

Please just read these few comments and give me a your idea.

I've tried to install python-occ with conda python 2.7 and it seems that all was fine. No error message during installation.
After this, I compiled pygear.py by python 2.7 and it worked with non error messages.

Then, I launched pygear but it returned me a message:

caelinux@Bruno:~/Documenti/pygear-0.24$ python /usr/local/lib/python2.7/dist-packages/pygear.py
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pygear.py", line 37, in <module>
    from OCC.TColgp import TColgp_Array1OfPnt2d, TColgp_Array1OfPnt
ImportError: No module named OCC.TColgp

Thank you very much for your help and ideas.

Bruno

tpaviot, 2017-07-27:
Which pythonocc version did you install ?

brunozilli, 2017-07-27:
caelinux@Bruno:/usr/local/lib/python2.7/dist-packages$ python
Python 2.7.13 (default, Jan 19 2017, 14:48:08) 
[GCC 6.3.0 20170118] on linux2

brunozilli, 2017-07-27:
sorry, pythonocc-018

I used this command line 

conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18

Anaconda 4.4.0 with python 2.7

Thank you

tpaviot, 2017-08-04:
@brunozilli try to import OCC.TColgp from a python prompt. btw, I don't know pygear and if it was updated to the latest pythonocc releases

tpaviot, 2017-12-01:
No news about this issue, please @brunozilli reopen if needed

closed 2017-12-01 by tpaviot

pedrogscruz, 2018-09-14:
Same issue here

hiro2016, 2018-12-26:
Same with anaconda python3.5 but I got a feeling maybe it's due to some outdated code I used.
```
Jupyter QtConsole 4.3.1
Python 3.5.6 |Anaconda, Inc.| (default, Aug 26 2018, 21:41:56) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.5.0 -- An enhanced Interactive Python. Type '?' for help.

from OCC.Core.gp import gp_Dir, gp_Ax2, gp_Circ, gp_Pnt
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-fc59bc42a444> in <module>()
----> 1 from OCC.Core.gp import gp_Dir, gp_Ax2, gp_Circ, gp_Pnt

ImportError: No module named 'OCC.Core' 
```

### The below works
```
from OCC.Display.SimpleGui import init_display
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
 
display, start_display, add_menu, add_function_to_menu = init_display()
my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()
 
display.DisplayShape(my_box, update=True)
start_display()
```

`conda activate v3.5;python filename_of_your_choice.py`

### Maybe irrelevant but the below sample code found at the official site does not work for python3 due to `operator.div` 
```
# http://www.pythonocc.org/category/quick-examples/
import operator

import OCC.GProp
import OCC.BRepGProp

import ifcopenshell
import ifcopenshell.geom

import numpy

# RGBA colors for the visualisation of elements
RED, GRAY = (1,0,0,1), (0.6, 0.6, 0.6, 0.1)

# Model freely available at:
# http://www.nibs.org/?page=bsa_commonbimfiles
ifc_file = ifcopenshell.open("Duplex_A_20110907_optimized.ifc")

# Settings to specify usage of pyOCC
settings = ifcopenshell.geom.settings()
settings.set(settings.USE_PYTHON_OPENCASCADE, True)

# Some helper functions to map to the list of walls
def create_shape(elem):
    return ifcopenshell.geom.create_shape(settings, elem)

def calc_volume(s):
    props = OCC.GProp.GProp_GProps()
    OCC.BRepGProp.brepgprop_VolumeProperties(s.geometry, props)
    return props.Mass()
    
def calc_area(s):
    props = OCC.GProp.GProp_GProps()
    OCC.BRepGProp.brepgprop_SurfaceProperties(s.geometry, props)
    return props.Mass()
    
def normalize(li):
    mean, std = numpy.mean(li), numpy.std(li)
[... 172 lines cut ...]
BRepExtrema.py           _GccEnt.so          _IncludeLibrary.so    _PrsMgr.so         StepElement.py       Visual3d.py
_BRepExtrema.so          GccGeo.py           __init__.py           __pycache__        _StepElement.so      _Visual3d.so
BRepFeat.py              _GccGeo.so          IntAna2d.py           Quantity.py        StepFEA.py           Visualization.py
_BRepFeat.so             GccInt.py           _IntAna2d.so          _Quantity.so       _StepFEA.so          _Visualization.so
BRepFilletAPI.py         _GccInt.so          IntAna.py             Resource.py        StepGeom.py          Voxel.py
_BRepFilletAPI.so        GccIter.py          _IntAna.so            _Resource.so       _StepGeom.so         _Voxel.so
BRepFill.py              _GccIter.so         IntCurve.py           RWStepAP203.py     StepRepr.py          XBRepMesh.py
_BRepFill.so             GCE2d.py            IntCurvesFace.py      _RWStepAP203.so    _StepRepr.so         _XBRepMesh.so
BRepGProp.py             _GCE2d.so           _IntCurvesFace.so     RWStepAP214.py     STEPSelections.py    XCAFApp.py
_BRepGProp.so            gce.py              _IntCurve.so          _RWStepAP214.so    _STEPSelections.so   _XCAFApp.so
BRepIntCurveSurface.py   _gce.so             IntCurveSurface.py    RWStepBasic.py     StepShape.py         XCAFDoc.py
_BRepIntCurveSurface.so  GCPnts.py           _IntCurveSurface.so   _RWStepBasic.so    _StepShape.so        _XCAFDoc.so
BRepLib.py               _GCPnts.so          InterfaceGraphic.py   RWStepDimTol.py    StepToGeom.py        XCAFPrs.py
_BRepLib.so              GC.py               _InterfaceGraphic.so  _RWStepDimTol.so   _StepToGeom.so       _XCAFPrs.so
BRepLProp.py             _GC.so              Interface.py          RWStepElement.py   StepToTopoDS.py      XSControl.py
_BRepLProp.so            Geom2dAdaptor.py    _Interface.so         _RWStepElement.so  _StepToTopoDS.so     _XSControl.so
BRepMAT2d.py             _Geom2dAdaptor.so   Intf.py               RWStepFEA.py       StepVisual.py
_BRepMAT2d.so            Geom2dAPI.py        _Intf.so              _RWStepFEA.so      _StepVisual.so
BRepMesh.py              _Geom2dAPI.so       IntImpParGen.py       RWStepGeom.py      StlAPI.py
```

I did the above trouble shooting out of the habit but I think I will use solid-python and openscad combination for the time being;they are sufficient for 3d printing small parts and easy to set up the env.    

Thanks for you great work tpaviot.  
Not for me right now but a cad library for python seems like a hard-to-write yet very important addition to the language.

Abhinav-Anil, 2020-02-19:
I'm using Python = 3.7
 PythonOCC = 0.18
Anaconda = 4.8.2

Traceback (most recent call last):
  File "occutils_geomplate.py", line 37, in <module>
    from OCC.GeomPlate import (GeomPlate_MakeApprox,
ModuleNotFoundError: No module named 'OCC.GeomPlate'

Please help and guide.