tpaviot/pythonocc-core #449: Realistic Rendering with STEP or STL

DEKHTIARJonathan, 2017-08-11:
Hello fellow PythonOCC Users,

I'm looking for a way to take render "realistically" any STEP or STL fed into the library.
Do you know if it is possible ? If not, do you know any open library that could help me ?

Thanks a lot for the help,

Jonathan DEKHTIAR

tpaviot, 2017-08-11:
Two solutions:

* use GLSL shaders, if you need realtime high quality rendering

* use a software like blender or 3dstudio. In this case, you need an exporter from pythonocc. You can have a look at occray (http://www.pythonocc.org/ecosystem/related_projects/occray/), which I think is a dead project but might be useful anyway: a povray based renderer for pythonocc.

Please let us know about your progresses regarding this field, this could be quite interesting.

tpaviot, 2017-08-11:
occray source code is now available at https://github.com/randfb/occray

DEKHTIARJonathan, 2017-08-11:
I will definetely try something. The code is really old, and absolutely not compatible with Python3.
I will try to update it, and if it works I may try to release it as PyPI package. That way, anyone will be able to install it quickly.

Please let this issue open, I will keep you updated ;)

Thanks again for your help, you rock !

jf---, 2017-08-11:
There's also this GPU physics based shader efforts by OpenCASCADE, called CADRays
see:

https://www.youtube.com/watch?v=eu_dMMQAhd0

DEKHTIARJonathan, 2017-08-11:
Damned @jf--- ! That's awesome :D 

I let here the url of the software: https://www.opencascade.com/content/cadrays

Do you know if there exists any API or Python Wrapper ?

jf---, 2017-08-29:
nope, I'd assume its a OCC product

tpaviot, 2017-09-19:
Raytracing rendering was added to the master branch, see PR #454 Although it cannot be considered as "realistic rendering", it's a step forward a "more realistic" rendering

closed 2017-09-19 by tpaviot