tpaviot/pythonocc-core #463: The function does not use parameters

kozintsev, 2017-10-22:
https://github.com/tpaviot/pythonocc-core/blob/30eade4e475dcac50386e36676036876e15e3141/examples/core_display_signal_slots.py#L72

Necessary:

def location_from_vector(x, y, z):
    trsf = gp_Trsf()
    trsf.SetTranslation(gp_Vec(x, y, z))
    loc = TopLoc_Location(trsf)
    return loc

tpaviot, 2017-10-22:
@kozintsev I can't understand your report, and don't see any error in core_display_signal_slots. What do you mean ?

kozintsev, 2017-10-22:
@tpaviot look at how you work function location_from_vector:
```python
def location_from_vector(x, y, z):
    trsf = gp_Trsf()
    trsf.SetTranslation(gp_Vec(500, 0, 0))
    loc = TopLoc_Location(trsf)
    return loc
```
The arguments of the function do not affect anything. If I set another value for X or Y or Z, nothing will change, the value will be equal to 500.
core_display_signal_slots works without errors, but there is a stylistic error.

tpaviot, 2017-10-22:
@kozintsev of course, I didn't see the mistake ;) Fixed by commit 0254ce9

closed 2017-10-22 by tpaviot