tpaviot/pythonocc-core #470: Extract geometrical entities from STEP file

jmauriciou, 2017-11-29:
Hello everybody!

I am starting with PythonOCC and OpenCascade Technology. I have a fundamental question. I have seen that it is possible to identify topological entities (such as vertices, edges, faces,...) from STEP files. I would like to know if it is possible to extract geometrical entities (such as a cylindrical surface, b_spline_surface) from a STEP file.

For instance I have the following STEP file: 

`#33=MANIFOLD_SOLID_BREP('',#34);
#34=CLOSED_SHELL('',(#61,#62,#63,#64,#65,#66,#67,#68,#69,#70,#71,#72,#73,
#74,#75,#76,#77,#78,#79,#80,#81,#82,#83));
#35=CYLINDRICAL_SURFACE('',#579,23.1283236048185);
#36=CYLINDRICAL_SURFACE('',#581,5.);
#37=CYLINDRICAL_SURFACE('',#591,5.);
#38=CYLINDRICAL_SURFACE('',#592,5.);
#39=CYLINDRICAL_SURFACE('',#593,5.);
#40=CYLINDRICAL_SURFACE('',#594,5.);
#41=FACE_OUTER_BOUND('',#109,.T.);
#42=FACE_OUTER_BOUND('',#110,.T.);`

I want to be able to extract all cylindrical surfaces (#35-#40) and their parameters (radius, length, coordinates of vertices) in python. Is this possible?  

Thanks in advance!

Jose

tpaviot, 2017-11-29:
@jmauriciou It's possible, yes.  The algorithm I have in mind would be:

* load the STEP file using the STEPImporter class

* traverse topology using the Topo class

* test each face to test wether it is a planar face, cylindrical etc.

Please post here a simple step file that could be used as an example.

jmauriciou, 2017-11-29:
Thanks for the reply. I am currently working on this simple part (stp file as txt). 

[image]

[part_example.txt](https://github.com/tpaviot/pythonocc-core/files/1514630/part_example.txt)

tpaviot, 2017-11-30:
@jmauriciou I just uploaded an example I built for you from the STEP file you provided. It's a very simple example that shows how to retrieve normal of a planra face, and axis of a cylindrical face (click any face from the graphical window, you'll get information about the face).

[image]

See commit 57ea7c332dae25a104552500380334c17ad49a5f. Feel free to improve this example and contribute back your progresses.

tpaviot, 2017-12-01:
Fixed.

closed 2017-12-01 by tpaviot

devanshu935, 2023-02-24:
Dear Thomas!
I am trying to execute the file with the following name from the  examples folder:
                      core_geometry_face_recognition_from_stepfile.py

I am not able to run it successfully even after a lot of effort. I have attached a screenshot of the error message I am getting in the Python console when I select a face in the 3D Viewer. I request you to help me with this. Thanks in advance.

[image]

Tanneguydv, 2023-02-28:
I have made a change in the example to fix it and created a merge request, 
you can look at the code [here](https://github.com/Tanneguydv/pythonocc-demos/blob/update-example/examples/core_geometry_face_recognition_from_stepfile.py) 

I guess that the selection mode is reset after displaying a new shape, so you have to had the function  ```SetSelectionModeFace()``` after the display of the shape.

[image]

devanshu935, 2023-02-28:
Dear Tanneguydv, thank you so very much for the help.
The code is working properly now
I'm really grateful for this. Had been stuck on it since last few days.
Thank you once again..

Nadjei74, 2024-04-04:
Hello Tanneguydv,
Please I am trying to access your code but the link is not working. Can you please work on that?