tpaviot/pythonocc-core #472: How to use Geom_BSplineSurface

tnakaicode, 2017-12-01:
I ran BSpline.py and got error folloing.

"RuntimeError: Standard_ConstructionError
Geom_BSplineSurface"

I don't know the cause of this error. Please tell me how to use Geom_BsplineSurface.

[BSpline.zip](https://github.com/tpaviot/pythonocc-core/files/1520740/BSpline.zip)

tpaviot, 2017-12-01:
@tnakaicode Please copy/paste python code in the issue tracker rather than a zip file

tpaviot, 2017-12-01:
Following resources about how to use Geom_BSpline_Surface:

* https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_geom___b_spline_surface.html

* https://cdn.rawgit.com/tpaviot/pythonocc-core/e05ec51b/doc/apidoc/0.18/OCC.Geom.html?highlight=geom_bsplinesurface#OCC.Geom.Geom_BSplineSurface

* https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_topology_face.py

closed 2017-12-01 by tpaviot