tpaviot/pythonocc-core #474: GeomAPI_PointsToBSplineSurface interpolate with point (0, 0, 0)

tnakaicode, 2017-12-02:
I used GeomAPI_PointsToBSplineSurface to make curved surface. But, my code added automatically point (0, 0, 0).

My code is following.

import math
import numpy as np

from OCC.gp import (gp_Pnt, gp_Sphere, gp_Ax3, gp_Dir, gp_Circ, gp_Ax2,
                    gp_Pnt2d, gp_Dir2d)
from OCC.BRepBuilderAPI import (BRepBuilderAPI_MakeEdge,
                                BRepBuilderAPI_MakeFace,
                                BRepBuilderAPI_MakeWire)
from OCC.TColgp import TColgp_Array2OfPnt
from OCC.GeomAPI import GeomAPI_PointsToBSplineSurface
from OCC.GeomAbs import GeomAbs_C2, GeomAbs_C0, GeomAbs_G1, GeomAbs_G2
from OCC.Geom2d import Geom2d_Line
from OCC.BRepLib import breplib_BuildCurves3d
from OCC.Quantity import Quantity_Color, Quantity_NOC_PINK

from OCC.Display.SimpleGui import init_display

def face(px, py, pz):
    nx, ny = px.shape
    pnt_2d = TColgp_Array2OfPnt(1, nx+1, 1, ny+1)
    for row in range (pnt_2d.LowerRow(), pnt_2d.UpperRow()):
        for col in range (pnt_2d.LowerCol(), pnt_2d.UpperCol()):
            i,j = row-1, col-1
            pnt = gp_Pnt (px[i, j], py[i, j], pz[i, j])
            pnt_2d.SetValue(row, col, pnt)
            print (i, j, px[i, j], py[i, j], pz[i, j])
    
    curve = GeomAPI_PointsToBSplineSurface(pnt_2d, 3, 8, GeomAbs_G2, 0.001).Surface()
    #surface = BRepBuilderAPI_MakeFace(curve, 1e-6)
    #return surface.Face()
    return curve
    
display, start_display, add_menu, add_function_to_menu = init_display()

px = np.linspace (-1, 1, 5) * 10
py = np.linspace (-1, 1, 5) * 10
mesh = np.meshgrid (px, py)
dat1 = mesh[0]**2 / (2*10) + mesh[1]**2 / (2*10)

surf1 = face(*mesh, dat1)
display.DisplayShape(surf1)
    
display.FitAll()
start_display()

tnakaicode, 2017-12-02:
Sorry. I can solve this problem.

```
def face(px, py, pz):
    nx, ny = px.shape
    pnt_2d = TColgp_Array2OfPnt(1, nx, 1, ny)
    print (pnt_2d.LowerRow(), pnt_2d.UpperRow())
    print (pnt_2d.LowerCol(), pnt_2d.UpperCol())
    for row in range (pnt_2d.LowerRow(), pnt_2d.UpperRow()+1):
        for col in range (pnt_2d.LowerCol(), pnt_2d.UpperCol()+1):
            i,j = row-1, col-1
            pnt = gp_Pnt (px[i, j], py[i, j], pz[i, j])
            pnt_2d.SetValue(row, col, pnt)
            print (row, col, i, j, px[i, j], py[i, j], pz[i, j])
    
    curve = GeomAPI_PointsToBSplineSurface(pnt_2d, 3, 8, GeomAbs_G2, 0.001).Surface()
    #surface = BRepBuilderAPI_MakeFace(curve, 1e-6)
    #return surface.Face()
    return curve
```

tpaviot, 2017-12-07:
Issue fixed, example heightmap added.

closed 2017-12-07 by tpaviot