tpaviot/pythonocc-core #480: Viewer3d object has no attribute SetRaytracingMode

frmdstryr, 2017-12-14:
I can't run the raytracing in the `core_display_raytraicng.py`  example. It opens fine but when I choose an option from the menu is exits with the given exception.

```
(occ) $:~/pythonocc-core/examples$ python core_display_raytracing.py 
creating bottle
bottle finished
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Xw_Window created.
Display3d class successfully initialized.
 ########################################
Traceback (most recent call last):
  File "core_display_raytracing.py", line 57, in raytracing_default_depth
    display.SetRaytracingMode()
AttributeError: 'Viewer3d' object has no attribute 'SetRaytracingMode'

```
Also getting similar errors with the `core_display_camera_projection.py` example:

`AttributeError: 'Viewer3d' object has no attribute 'SetAnaglyphMode'`

Versions:
- Linux xps 4.10.0-37-generic #41~16.04.1-Ubuntu SMP Fri Oct 6 22:42:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
- Python 3.6.3 | packaged by conda-forge | (default, Dec  9 2017, 16:18:26) 
- PyQt5 (5.9.2)
>>> import OCC
>>> OCC.VERSION
'0.18'

I installed everything with conda today.

Thank you!

tpaviot, 2017-12-14:
@frmdstryr raytracing is not available from the 0.18 release. It's a recent addition, about to be released in the coming 0.18.1 version. Try a nightly build or wait for a couple of days before the next release is out.

closed 2017-12-14 by tpaviot

frmdstryr, 2017-12-14:
Sorry about that. Will try the nightly build. Thanks for the prompt reply!