tpaviot/pythonocc-core #484: Error windows 1 - miniconda3 64bit

bjaraujo, 2017-12-25:
I get the following error trying to run:

from OCC.Display.SimpleGui import init_display
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
 
display, start_display, add_menu, add_function_to_menu = init_display()
my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()
 
display.DisplayShape(my_box, update=True)
start_display()

Traceback (most recent call last):
  File "block.py", line 1, in <module>
    from OCC.Display.SimpleGui import init_display
  File "c:\ProgramData\Miniconda3\lib\site-packages\OCC\Display\SimpleGui.py", line 26, in <module>
    from OCC.Display.OCCViewer import OffscreenRenderer
  File "c:\ProgramData\Miniconda3\lib\site-packages\OCC\Display\OCCViewer.py", line 41, in <module>
    from OCC.Visualization import Display3d
  File "c:\ProgramData\Miniconda3\lib\site-packages\OCC\Visualization.py", line 21, in <module>
    _Visualization = swig_import_helper()
  File "c:\ProgramData\Miniconda3\lib\site-packages\OCC\Visualization.py", line 20, in swig_import_helper
    return importlib.import_module('_Visualization')
  File "c:\ProgramData\Miniconda3\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_Visualization'

tpaviot, 2018-01-25:
@bjaraujo is is still failing ?

bjaraujo, 2018-01-25:
I haven't checked recently. I will probably try again on the weekend.

bjaraujo, 2018-03-25:
Should I install pythonocc-core==0.18.1 ?

bjaraujo, 2018-03-25:
It gives during install:

UnsatisfiableError: The following specifications were found to be in conflict:
  - pythonocc-core==0.18.1 -> freetype==2.7
  - scikit-image -> imageio[version='>=2.1.0'] -> pillow -> freetype[version='>=2.8,<2.9.0a0']
Use "conda info <package>" to see the dependencies for each package.

bjaraujo, 2018-03-25:
Still happens:

Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\OCC\Visualization.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\ProgramData\Anaconda3\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: DLL load failed: The specified procedure could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "simple.py", line 1, in <module>
    from OCC.Display.SimpleGui import init_display
  File "C:\ProgramData\Anaconda3\lib\site-packages\OCC\Display\SimpleGui.py", line 26, in <module>
    from OCC.Display.OCCViewer import OffscreenRenderer
  File "C:\ProgramData\Anaconda3\lib\site-packages\OCC\Display\OCCViewer.py", line 41, in <module>
    from OCC.Visualization import Display3d
  File "C:\ProgramData\Anaconda3\lib\site-packages\OCC\Visualization.py", line 21, in <module>
    _Visualization = swig_import_helper()
  File "C:\ProgramData\Anaconda3\lib\site-packages\OCC\Visualization.py", line 20, in swig_import_helper
    return importlib.import_module('_Visualization')
  File "C:\ProgramData\Anaconda3\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_Visualization'

tpaviot, 2019-01-13:
Fixed in current master.

hypotenuse345, 2019-05-10:
Sorry to bother. I seem to encounter the same issue. Have you solved this problem yet?
My pythonocc version is latest(?) 0.18.2. Is this the problem of my computer itself, like missing some toolkits?
Here is the error report:
`>>> import OCC.BRep
D:\Anaconda\envs\py36\lib\site-packages\OCC\BRep.py:3: DeprecationWarning: OCC.BRep is deprecated since pythonocc-0.18.2. Use OCC.Core.BRep
  warnings.warn("OCC.BRep is deprecated since pythonocc-0.18.2. Use OCC.Core.BRep", DeprecationWarning)
Traceback (most recent call last):
  File "D:\Anaconda\envs\py36\lib\site-packages\OCC\Core\BRep.py", line 34, in swig_import_helper
    return importlib.import_module(mname)
  File "D:\Anaconda\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 648, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 560, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
ImportError: DLL load failed: The specified procedure could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\Anaconda\envs\py36\lib\site-packages\OCC\BRep.py", line 5, in <module>
    from OCC.Core.BRep import *
  File "D:\Anaconda\envs\py36\lib\site-packages\OCC\Core\BRep.py", line 37, in <module>
    _BRep = swig_import_helper()
  File "D:\Anaconda\envs\py36\lib\site-packages\OCC\Core\BRep.py", line 36, in swig_import_helper
    return importlib.import_module('_BRep')
  File "D:\Anaconda\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_BRep'
>>>`

Zebrafish007, 2020-01-02:
Same issue with `conda install -c dlr-sc -c pythonocc pythonocc-core=7.4.0rc1` version

      File "X:\Y\Anac\envs\py36\lib\site-packages\OCC\Core\Visualization.py", line 13, in <module>
        from . import _Visualization
    ImportError: DLL load failed: The specified procedure could not be found.

rainman110, 2020-01-02:
@Zebrafish007 I can't reproduce your issue. Can you please provide a list of all conda packages installed?

I have the following packages installed:
```
(pythonocctest) C:\>conda list
# packages in environment at C:\Users\sigg_ma\.conda\envs\pythonocctest:
#
# Name                    Version                   Build  Channel
certifi                   2019.11.28               py36_0
freeimageplus             3.18.0               h6538335_2    dlr-sc
freetype                  2.6.5                hfa6e2cd_4    dlr-sc
opencascade               7.4.0                he980bc4_4    dlr-sc
pip                       19.3.1                   py36_0
python                    3.6.9                h5500b2f_0
pythonocc-core            7.4.0rc1         py36he980bc4_0    dlr-sc
setuptools                42.0.2                   py36_0
six                       1.13.0                   py36_0
sqlite                    3.30.1               he774522_0
tbb                       2019.5               he980bc4_0    dlr-sc
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.16.27012          hf0eaf9b_1
wheel                     0.33.6                   py36_0
wincertstore              0.2              py36h7fe50ca_0
```

Importing ```OCC.Core.Visualization``` works here without any problem

Zebrafish007, 2020-01-02:
certifi                      2019.11.28       py36_0
freeimageplus         3.18.0               h6538335_2    dlr-sc
freetype                  2.9.1                ha9979f8_1
opencascade           7.4.0                h811d2cf_1    dlr-sc
pip                         19.3.1                   py36_0
python                   3.6.9                    h5500b2f_0
pythonocc-core      7.4.0rc1               py36he980bc4_0    dlr-sc
pythreejs                2.1.1                    pypi_0    pypi
setuptools              42.0.2                  py36_0
six                         1.13.0                   py36_0
sqlite                     3.30.1                   he774522_0
tbb                       2019.4                   h74a9793_0
vc                         14.1                       h0510ff6_4
vs2015_runtime    14.16.27012          hf0eaf9b_1
wheel                     0.33.6                   py36_0
wincertstore              0.2              py36h7fe50ca_0

..hmm.. same brew.. could have copied yours ;-)

Zebrafish007, 2020-01-02:
Strangely enough the following works for me too...

(py36) X:\Y\USERNAME>python
    Python 3.6.9 |Anaconda, Inc.| (default, Jul 30 2019, 14:00:49) [MSC v.1915 64 bit (AMD64)] on win32
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import OCC.Core.Visualization
    >>>

rainman110, 2020-01-02:
What command does not work?

Zebrafish007, 2020-01-02:
The import works from cmdline..but not from within Komodo Edit when I run the following script:

from cadquery import Workplane
from ipywidgets import embed
from jupyter_cadquery.cad_view import CadqueryView
from jupyter_cadquery.cadquery import Assembly
from jupyter_cadquery.cadquery import Part

# Create a simple assembly
box1 = Workplane('XY').box(10, 10, 10).translate((0, 0, 5))
a1 = Assembly([Part(box1)], "example 1")

# Generate HTML
a1.collect_shapes()
view = CadqueryView()
for shape in a1.collect_shapes():
    view.add_shape(shape["name"], shape["shape"], shape["color"])
renderer = view.render()
embed.embed_minimal_html('export.html', views=renderer, title='Renderer')

renderer

Zebrafish007, 2020-01-02:
The import works from cmdline..but not from within Komodo Edit when I run the following script

    from cadquery import Workplane
    from ipywidgets import embed
    from jupyter_cadquery.cad_view import CadqueryView
    from jupyter_cadquery.cadquery import Assembly
    from jupyter_cadquery.cadquery import Part

    # Create a simple assembly
    box1 = Workplane('XY').box(10, 10, 10).translate((0, 0, 5))
    a1 = Assembly([Part(box1)], "example 1")

    # Generate HTML
    a1.collect_shapes()
    view = CadqueryView()
    for shape in a1.collect_shapes():
        view.add_shape(shape["name"], shape["shape"], shape["color"])
    renderer = view.render()
    embed.embed_minimal_html('export.html', views=renderer, title='Renderer')

    renderer

rainman110, 2020-01-02:
To figure out, whether pythonocc has an issue, we need a minimal example **without** cadquery that demonstrates the issue. Here, we also have jupyter, cadquery etc that we not in your package list.

There might be some dll collisions, that why I need to know, which packages you have installed. Whether cadquery or juypter installs e.g. another freetype are important infos.

Zebrafish007, 2020-01-02:
This requires to have freeCAD and pythreejs installed too. When you hit:

     Traceback (most recent call last):
       File "X:\Y\anac\lib\site-packages\cadquery\freecad_impl\__init__.py", line 152, in <module>
        import FreeCAD
    ModuleNotFoundError: No module named 'FreeCAD'

You have to add under:

    elif sys.platform.startswith('win'):
        # Try all the usual suspects
        for _PATH in [
                "c:/Program Files/FreeCAD0.12/bin",
                ... snippet ...
                your freeCAD installation folder.      (in the /bin)

Komodo edit = 11.1

Zebrafish007, 2020-01-02:
okay..what elso you need to know?

rainman110, 2020-01-02:
The best would be a list of all DLLs from the env/bin folder of your conda environment and also the version numbers of the DLLs.

The best tool for analyzing DLL issue is Dependency Walker. It is a pretty old tool but does the job. Here first add the env/bin folder to the search path of dependency walker. Than drag in the _Visualization.pyd from env/Lib/site-packages/OCC/Core/ into the dependency walker.

Dependency Walker will tell you all DLL related problems.

Main  source of problems are:
 - Missing DLLs
 - Wrong DLLs (e.g. a wrong DLL Version)
 - Missing functions in a DLL

Zebrafish007, 2020-01-02:
There is no (py36) x:\y\Anac\envs\py36\bin. There is a x:\y\Anac\envs\py36\DLLs.

rainman110, 2020-01-02:
Right, the DLLs are probably in
```x:\y\Anac\envs\py36\Library\bin```

Zebrafish007, 2020-01-02:
Yep... found it but there is no _Visualization in that folder. Its located at: X:\Y\Anac\envs\py36\Lib\site-packages\OCC\Core. Copied the file  _Visualization.pyd to x:\y\Anac\envs\py36\Library\bin failed to fix it.

rainman110, 2020-01-02:
Sure... ! Just look in my reply above. The _Visualization.pyd is inside env\Lib\site-packages\OCC\Core and MUST remain there!

Zebrafish007, 2020-01-02:
I've opend DepWalk and did open the file _Visualization.pyd via selection X:\Y\Anac\envs\py36\Lib\site-packages\OCC\Core and I'm getting the (not Responding) at the dialogbar.

Zebrafish007, 2020-01-02:
I've opend DepWalk and did open the file _Visualization.pyd via selection X:\Y\Anac\envs\py36\Lib\site-packages\OCC\Core and I'm getting the (not Responding) at the dialogbar. Happens also when I explicitly run it as admin.

Zebrafish007, 2020-01-02:
I've opend DepWalk and did open the file _Visualization.pyd via selection X:\Y\Anac\envs\py36\Lib\site-packages\OCC\Core and I'm getting the (not Responding) at the dialogbar. Happens also when I explicitly run it as admin under win10 64 pro. Changed to run as windows 7 and combo with admin... its busy... and nolonger (not responding).

Zebrafish007, 2020-01-02:
I've opend DepWalk and did open the file _Visualization.pyd via selection X:\Y\Anac\envs\py36\Lib\site-packages\OCC\Core and I'm getting the (not Responding) at the dialogbar. Happens also when I explicitly run it as admin under win10 64 pro. Changed to run as windows 7 and combo with admin... its busy... and nolonger (not responding) but in task manager... it hangs with the (not responding) message.

rainman110, 2020-01-02:
Then use the following program instead:
https://github.com/lucasg/Dependencies

(I also had the same issue with Dependency Walker occasionally)

Zebrafish007, 2020-01-02:
I don't see the dependencyGUI.exe... do I have to build it myself?

Zebrafish007, 2020-01-02:
Dependencies works. It gives me:

x:\y\Anac\Library\bin\TKOpenGl.dll
python36.dll
x:\y\Anac\Library\bin\TKPrim.dll
x:\y\Anac\Library\bin\TKV3d.dll
x:\y\Anac\Library\bin\TKMesh.dll
x:\y\Anac\Library\bin\TKService.dll
x:\y\Anac\Library\bin\TKTopAlgo.dll
x:\y\Anac\Library\bin\TKBRep.dll
x:\y\Anac\Library\bin\TKMath.dll
x:\y\Anac\Library\bin\TKernel.dll
C:\WINDOWS\system32\user32.dll
C:\WINDOWS\system32\MSVCP140.dll
C:\WINDOWS\system32\VCRUNTIME140.dll
api-ms-win-crt-runtime-l1-1-0.dll -> C:\WINDOWS\system32\ucrtbase.dll
api-ms-win-crt-math-l1-1-0.dll -> C:\WINDOWS\system32\ucrtbase.dll
api-ms-win-crt-stdio-l1-1-0.dll -> C:\WINDOWS\system32\ucrtbase.dll
api-ms-win-crt-string-l1-1-0.dll -> C:\WINDOWS\system32\ucrtbase.dll
api-ms-win-crt-heap-l1-1-0.dll -> C:\WINDOWS\system32\ucrtbase.dll
api-ms-win-crt-filesystem-l1-1-0.dll -> C:\WINDOWS\system32\ucrtbase.dll
C:\WINDOWS\system32\kernel32.dll
C:\WINDOWS\system32\OPENGL32.dll
C:\WINDOWS\system32\gdi32.dll
api-ms-win-crt-convert-l1-1-0.dll -> C:\WINDOWS\system32\ucrtbase.dll

and python36.dll gives a questionmark. This should be found under env/py36 but it isn't.

rainman110, 2020-01-02:
This looks fine. Then Freecad could be the problem. You said, that you are automatically adding Freecad to the path.

Could you prepend the FreeCad\bin directory to PATH and test again, if importing Visualization still works.

Zebrafish007, 2020-01-02:
Adding additional folder for python36.dll fixed that questionmark.

Zebrafish007, 2020-01-02:
Did add Z:\AA\FreeCAD\bin to envirmental variables and keep getting:

Traceback (most recent call last):
  File "A:\B\test4.py", line 3, in <module>
    from jupyter_cadquery.cad_view import CadqueryView
  File "X:\Y\Anac\envs\py36\lib\site-packages\jupyter_cadquery\__init__.py", line 18, in <module>
    from .cad_display import set_sidecar
  File "X:\Y\Anac\envs\py36\lib\site-packages\jupyter_cadquery\cad_display.py", line 25, in <module>
    from .cad_view import CadqueryView
  File "X:\Y\Anac\envs\py36\lib\site-packages\jupyter_cadquery\cad_view.py", line 30, in <module>
    from OCC.Core.Visualization import Tesselator
  File "X:\Y\Anac\envs\py36\lib\site-packages\OCC\Core\Visualization.py", line 16, in <module>
    from . import _Visualization
ImportError: DLL load failed: The specified procedure could not be found.

rainman110, 2020-01-02:
Please test the other way around.

 - Go to console into your conda env
 - Add FreeCad\bin to env paths
 - open python.exe
 - try importing only OCC.Core.Visualization

We are searching here for a MINIMAL example that demonstrates the issue. CadQuery and Jupyter should stay out of the loop if possible.

rainman110, 2020-01-02:
I could imagine, that FreeCad uses different OpenCASCADE DLLs that are located in FreeCad\bin (i.e. a different opencascade version). The mixing of two different OCCT Versions will definitely lead to problems.

Zebrafish007, 2020-01-02:
..as mentioned.... from the commandline (py36) python > import OCC.Core.Visualization works.

Zebrafish007, 2020-01-02:
..as mentioned.... from the commandline (py36) python > import OCC.Core.Visualization works. Its when I run the cadquery script from within Komodo edit... it fails to find _visualization.pyd.

Zebrafish007, 2020-01-02:
..as mentioned.... from the commandline (py36) python > import OCC.Core.Visualization works. Its when I run the cadquery script from within Komodo edit... it fails to find _visualization.pyd. SO my suggestion would be at the import statement include a custom... which I tried to implement but failed thus far.

Zebrafish007, 2020-01-02:
In the folder freeCAD I don't see anything remotle named OCC or opencascade.

Zebrafish007, 2020-01-02:
In the folder freeCAD I don't see anything remotly named OCC or opencascade.

rainman110, 2020-01-02:
Inside the FreeCad\bin directory, there are many TK*.dll files. All belong to opencascade.

Freecad 0.18 uses opencascade 6.7.0. The new pythonocc uses 7.4.0. Hence, the DLLs are incompatible and therefore you get the error.

@tpaviot This issue is not pythonocc related and more an issue of cadquery (using freecad) mixing with pythonocc in a non-safe manner.

tpaviot, 2020-01-02:
@rainman110 Yes I read your discussion. I thought freecad has recently moved to occt7.3.0, I might be wrong.

@Zebrafish007 I don't know, on windows, how to link to a specific library by setting its path. If you need to use concurrent versions of occt, I suggest exchanging shape serializations between versions. It's not elegant but it will work.

rainman110, 2020-01-02:
@tpaviot Yes, the current development version are using OCCT 7.3.0. This is still incompatible to 7.4.0, so not a good idea to simply mix both libs together.

Zebrafish007, 2020-01-02:
..above my paygrade I guess. Different route: conda install -c conda-forge -c cadquery cadquery=2 is tossing the below error:

Different route: conda install -c conda-forge -c cadquery cadquery=2 is tossing the below error:

Traceback (most recent call last):
  File "Z:AA\test4.py", line 1, in <module>
    from cadquery import Workplane
  File "C:\X\Y\Anac\envs\py36\lib\site-packages\cadquery\__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix
  File "C:\X\Y\Anac\envs\py36\lib\site-packages\cadquery\occ_impl\geom.py", line 3, in <module>
    from OCC.Core.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp, gp_XYZ
  File "C:\X\Y\Anac\envs\py36\lib\site-packages\OCC\Core\gp.py", line 36, in <module>
    _gp = swig_import_helper()
  File "C:\X\Y\Anac\envs\py36\lib\site-packages\OCC\Core\gp.py", line 35, in swig_import_helper
    return importlib.import_module('_gp')
  File "C:\X\Y\Anac\envs\py36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_gp'

(py36) C:\A\B>conda list
# packages in environment at c:\Python\Anac\envs\py36:
#
# Name                    Version                   Build  Channel
_anaconda_depends         2019.03                  py36_0
cadquery                  2.0RC1                    py3.6    cadquery
certifi                   2019.11.28               py36_0    conda-forge
freeimageplus             3.17.0                        0    cadquery
freetype                  2.9.1                ha9979f8_1
jupyter-cadquery          0.9.2rc5                 pypi_0    pypi
opencascade               7.4.0                h811d2cf_1    dlr-sc
oce                       0.18.2                        0    cadquery
opencascade               7.4.0                he980bc4_4    dlr-sc  # after delete OCC and reinstal.
openssl                   1.1.1d               hfa6e2cd_0    conda-forge
pip                       19.3.1                   py36_0
python                    3.6.9                h5500b2f_0
pythonocc-core            0.18.2          py36_vc14h39e3cac_0  [vc14]  cadquery
pythreejs                 2.1.1                    pypi_0    pypi
setuptools                42.0.2                   py36_0
six                       1.13.0                   py36_0
sqlite                    3.30.1               he774522_0
tbb                       2019.4               h74a9793_0
tbb                       2019.5               he980bc4_0    dlr-sc  # after delete OCC and reinstal.
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.16.27012          hf0eaf9b_1
wheel                     0.33.6                   py36_0
wincertstore              0.2              py36h7fe50ca_0

rainman110, 2020-01-02:
There you are
> opencascade 7.4.0 he980bc4_4 dlr-sc # after delete OCC and reinstall.

You have both oce (opencascade community edition) and opencascade which does not sound very healthy and is very likely to break.

Unfortunately, it is not yet possible to define package conflicts in conda, see  https://github.com/conda/conda/issues/608

rainman110, 2020-01-02:
By the way... When I try to install cadquery, I get an error due to conflicts:

```
(pythonocctest) C:\>conda install -c conda-forge -c cadquery cadquery=2
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: /
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining pythonocc-core:  43%|██████████████████████▉                              | 73/169 [00:00<00:00, 1556.76it/s]
Comparing specs that have this dependency:   0%|                                                 | 0/2 [00:00<?, ?it/s]

Finding conflict paths:   0%|                                                                    | 0/2 [00:00<?, ?it/s]

Finding shortest conflict path for pythonocc-core=0.18.2:   0%|                                  | 0/2 [00:00<?, ?it/s]

Finding shortest conflict path for pythonocc-core=7.4.0rc1:  50%|████████████████                | 1/2 [00:00<?, ?it/|

Comparing specs that have this dependency:  50%|████████████████████▌                    | 1/2 [00:01<00:01,  1.02s/it]

Finding conflict paths:   0%|                                                                    | 0/1 [00:00<?, ?it/s]

Finding shortest conflict path for pythonocc-core=7.4.0rc1:   0%|                                | 0/1 [00:00<?, ?it/s]

failed

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (pythonocc-core):

  - cadquery=2 -> pythonocc-core=0.18.2

The following specifications were found to be incompatible with each other:

Package pythonocc-core conflicts for:
pythonocc-core=7.4.0rc1
```

So conda already notices, that this is not going to work.

tpaviot, 2020-01-02:
Yes, absolutely, impossible to use conda packges that rely on different versions of dynamic libraries. One or the other downgrade/upgrade the versions according to its requirements.

Zebrafish007, 2020-01-04:
So what approach on how to proceed would then be the best...? drop everything below cadquery 2.0? And if so get pythonocc 7.4 aligned in a 7.5 version with it?

adam-urbanczyk, 2020-01-04:
@Zebrafish007  what are you trying to achieve? If you want to run CQ with OCCT7.4 it is currently not possible.If you are having issues installing CQ please follow the instructions from the readme. If it does not work, feel free to open an issue in the CQ repo.

Zebrafish007, 2020-01-04:
Both methods of installation end-up in throwing tracebacks at me. The first one is tackeld to be an CQ issue with mixing libraries and the second ends with pythonocc bumping on missing module as mentioned above:

     File "C:\X\Y\Anac\envs\py36\lib\site-packages\OCC\Core\gp.py", line 35, in swig_import_helper
return importlib.import_module('gp')
     File "C:\X\Y\Anac\envs\py36\lib\importlib_init.py", line 126, in import_module
     return _bootstrap._gcd_import(name[level:], package, level)
     ModuleNotFoundError: No module named '_gp'

SO either way... I got stuck on installation/incompatibility issues and want to know which route does produce neither.

rainman110, 2020-01-04:
Simply install cadquery with the old version of pythonocc.

adam-urbanczyk, 2020-01-04:
This is the advised way: `conda install -c conda-forge -c cadquery cadquery=2`

Zebrafish007, 2020-01-06:
The `conda install -c conda-forge -c cadquery cadquery=2` throws me a import '_gp' error as mentioned on 2nd and 4th of January on pythonocc.