tpaviot/pythonocc-core #485: python version bump

jkbgbr, 2017-12-27:
Dear all,

as this is my first comment here so let me tell how grateful I am for the work of the package maintainers. As someone quite illiterate in C/C++ the python bindings are for me THE way to use OCC. Thank you for your efforts!

According to the README.md python version 3.5 is needed, but later in a [comment](https://github.com/tpaviot/pythonocc-core/issues/405#issuecomment-284138437) it is stated that 3.6 is the version to be used.
I wonder if this could be fixed. 
A slightly related question: until now I used the conda package, but I'm planning to compile from source (under Windows), as anaconda is quite bloated for my needs. Is the INSTALL.md file up-to-date?

Cheers,
Gabor

tpaviot, 2017-12-27:
pythonocc supports python 2.7 and python3k (3.3, 3.4, 3.5 and 3.6). The README.md does not explicitely says that python 3.6 is *required*, it's just an example.

Under Windows, I'll update the INSTALL.md, it maybe outdated.

jkbgbr, 2017-12-27:
I see, thanks for the information. 
I have some difficulties when trying to run the command included at [the conda package page] (https://anaconda.org/pythonocc/pythonocc-core) when installing on a mint anaconda, that's why I asked if it was a python version problem.
I'll keep an eye on the new version of the install and try when it is updated.