tpaviot/pythonocc-core #487: Trying to install review/tess branch on jupyter

bjaraujo, 2017-12-27:
!conda install --yes -c tpaviot/label/cd-review/tess pythonocc-core
Fetching package metadata ....
WARNING: The remote server could not find the noarch directory for the
requested channel with url: https://conda.anaconda.org/tpaviot/label/cd-review/tess

It is possible you have given conda an invalid channel. Please double-check
your conda configuration using `conda config --show`.

If the requested url is in fact a valid conda channel, please request that the
channel administrator create `noarch/repodata.json` and associated
`noarch/repodata.json.bz2` files, even if `noarch/repodata.json` is empty.
$ mkdir noarch
$ echo '{}' > noarch/repodata.json
$ bzip2 -k noarch/repodata.json
.............

PackageNotFoundError: Packages missing in current channels:
            
  - pythonocc-core

tpaviot, 2017-12-27:
what's your os ? platform ?

bjaraujo, 2017-12-28:
Initially, I was using the online version: https://try.jupyter.org/
I also tried jupyter that comes with Anaconda3 on Windows 10 and it gave me same result.

Would like to try pythonocc on jupyter. It seems very promising!

Thanks for pythonocc btw.

tpaviot, 2019-12-29:
Have a look at https://mybinder.org/v2/gh/tpaviot/pythonocc-binderhub/master

And you can download/install latest version 7.4.0beta2 to check again. Reopen the ticket if you still face this issue.

closed 2019-12-29 by tpaviot