tpaviot/pythonocc-core #488: running your quick example

wzbernstein, 2017-12-28:
I was trying to run the quick example (http://www.pythonocc.org/category/quick-examples/) on a Python 3.5 environment.  I tracked down the example model off of NIBS (http://www.nibs.org/?page=bsa_commonbimfiles).  When I ran the code, it returned an error (below).  Any ideas?  Thanks.

runfile('C:/Users/wzb/.spyder-py3/temp.py', wdir='C:/Users/wzb/.spyder-py3')
Traceback (most recent call last):

  File "<ipython-input-6-69db953d5903>", line 1, in <module>
    runfile('C:/Users/wzb/.spyder-py3/temp.py', wdir='C:/Users/wzb/.spyder-py3')

  File "C:\Users\wzb\AppData\Local\Continuum\anaconda3\envs\py35\lib\site-packages\spyder\utils\site\sitecustomize.py", line 710, in runfile
    execfile(filename, namespace)

  File "C:\Users\wzb\AppData\Local\Continuum\anaconda3\envs\py35\lib\site-packages\spyder\utils\site\sitecustomize.py", line 101, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/wzb/.spyder-py3/temp.py", line 49, in <module>
    feature = normalize(map(operator.div, areas, volumes))

AttributeError: module 'operator' has no attribute 'div'

wzbernstein, 2017-12-28:
Is your library only available for 2.x?

wzbernstein, 2017-12-29:
The quick example works using ipcopenshell (for 2.7 win64) and pythonocc in my py27 env.

tpaviot, 2017-12-29:
pythonocc supports both python 2.7 and 3.x

thanks for the feedback regarding this quick example published a long time ago !

closed 2017-12-29 by tpaviot

wzbernstein, 2017-12-29:
Thanks for the message.  I will try again.  The STEP read example seems to work well also with my own models.