tpaviot/pythonocc-core #489: Standard_OutOfRange

FullStackRafik, 2017-12-29:
Hi,
I get the following error when reading AP238 file:
Failed to read the file: Standard_OutOfRange.
Can you please tell me if there is any solution for that.

Thanks

tpaviot, 2017-12-29:
AP238 is not supported. Only AP203 and 214.

FullStackRafik, 2017-12-29:
Hey tpaviot,
Thanks for your reply.
Is there any solution for that? or any other library that can support AP238?
Thanks

tpaviot, 2017-12-29:
Try stepcode https://github.com/stepcode/stepcode

FullStackRafik, 2017-12-29:
Thanks again, I'll try to use it.
Is there any support for python in this library?

tpaviot, 2018-01-01:
Using stepcode, you can parse an EXPRESS file and generate a python module, then it's possible to load/write files accordingly. I wrote the first version of this exporter, years ago, I don't know the current status.

closed 2018-01-01 by tpaviot

FullStackRafik, 2018-01-01:
Thanks for your reply.
I know how to use the parser to generate python from express, but I need to know how to display the 3D view of AP238.