tpaviot/pythonocc-core #490: Generation of swig files from oce not consistent

tommy3001, 2018-01-01:
I am trying to generate the SWIG files with the pythonocc-generator based on the github master brunch of oce. 
But after that I am not able anymore to build pythonocc-core (based on master brunch after swig file generation with python2 generate_wrapper.py). Build process cancelled with 

include/oce/Quantity_Color_1.hxx:20:8:
static Quantity_Parameter TheEpsilon = 0.0001;

Quantity_Parameter is not a type

What is the base for generation the swig files? It seems to be that the generated swig files are different then the swig files in the pythonocc-core master brunch repository.

My first aim was only to update and align the API Documentation of pythonocc-core to use it to extend the example documenation.....

tpaviot, 2018-01-01:
Using latest oce master and pythonocc-generator master, you should generate the SWIG files of the pythonocc-core master branch. They should be *exactly* the same, that is to say that the diff file is empty.

Can you please post the diff ?

Note : I run the pythonocc-generator on Windows, on Linux the hxx header may be sorted differently, but I need to see the diff

tommy3001, 2018-01-02:
[diff_swigfiles.txt](https://github.com/tpaviot/pythonocc-core/files/1598888/diff_swigfiles.txt)

It is a very huge file.

The command was diff GeneratedFiles OriginalFilesFromMaster

Both, oce and pythonocc-core were cloned on a fresh linux system from github, master brunch

I got now a new issue during the build process of pythonocc-core after generating the swig files with pythonocc-generator:

[ 95%] Built target _TopClass
In file included from /root/pythonocc-core/cmake-build/Unix/x86_64-MinSizeRel-64/BRepMeshPYTHON_wrap.cxx:4227:0:
/usr/local/lib/oce-0.18-dev/../../include/oce/BRepMesh_Vertex.hxx: In member function ‘Standard_Boolean BRepMesh_Vertex::IsEqual(const BRepMesh_Vertex&) const’:
/usr/local/lib/oce-0.18-dev/../../include/oce/BRepMesh_Vertex.hxx:125:41: error: ‘Precision’ has not been declared
     return (myUV.IsEqual(theOther.myUV, Precision::PConfusion()));
                                         ^~~~~~~~~
make[2]: *** [CMakeFiles/_BRepMesh.dir/build.make:72: CMakeFiles/_BRepMesh.dir/Unix/x86_64-MinSizeRel-64/BRepMeshPYTHON_wrap.cxx.o] Error 1

tommy3001, 2018-01-04:
That the generated SWIG files are changing its content in dependence of the platform is not such a big problem for me now. The problem with the missing headers caused by the include order can be handled by myself. I will make a small pull request regarding the header issue.

tommy3001, 2018-01-05:
I added a newline character to the wrapper generator. ;-) 
From my view we can close this here. I would prefere when the updated swig files of this repo would be also updated. But this should be from your system....

closed 2018-01-08 by tommy3001