tpaviot/pythonocc-core #491: brepgprop_LinearProperties crashes

seeul8er, 2018-01-03:
I try to get the length of an TopoDS_Edge or TopoDS_Wire using:
```python
props = GProp_GProps()
brepgprop_LinearProperties(_wire, props)
print(props.Mass())
```
Where `_wire` is a `TopoDS_Wire`.
The code/programm crashes with: `Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)`

Background: I imported a file as .stp or .stl and got the edges etc. 
The code seems to crash on `brepgprop_LinearProperties()`
Used `pythonocc-core 0.18.1`

Amazing project btw.! :+1:

tpaviot, 2018-01-12:
can you please provide a test case. Hard to debug without any additional material

tpaviot, 2019-12-29:
Try with the latest version, if ever it still fails reopen the ticket

```
conda create --name=whatever-name python=3.7
source activate whatever-name
conda install -c dlr-sc -c pythonocc pythonocc-core=7.4.0beta2
```

closed 2019-12-29 by tpaviot