tpaviot/pythonocc-core #495: Make Ellipsoid

BenjaminMartin1986, 2018-01-16:
Hello, 

Is it possible to make an 3d-elliptic cylinder base (cylinder with elliptic base) using PythonOCC ? I'm currently trying with BRepPrimAPI librairy but I just fail ... ?
Thanks !
Benjamin

tpaviot, 2018-01-16:
There's no ellipsoid in the BRepPrimAPI module, only Sphere. 2 options:

* create a Sphere using BRepPrimAPI_MakeSphere, and scale the shape in either x, y or z directions ;

* create a 2d ellipse (using gp_*), rotate this ellipse to create a Face, then fill the Face to make a Solid.

I'd choose option 1.

BenjaminMartin1986, 2018-01-16:
Thank you for answer. I'm trying option 1 (I think it is BRepPrimAPI_MakeCylinder instead of BRepPrimAPI_MakeSphere). Is it possible to scale using gp_Trsf().SetScale in just 2 directions ?
Thanks !