tpaviot/pythonocc-core #502: Proper meshing options

MagaFem, 2018-01-28:
I'm having trouble finding the proper options for meshing a 3d volume. @tpaviot Possible to share your code of you last Twitter posts?

https://twitter.com/pythonocc/status/952089110914756609

https://twitter.com/pythonocc/status/876728963460202496

tpaviot, 2018-01-31:
The screenshots you mention show a 2d triangular surfacic mesh. The second one is obtained using the MEFISTO2 Smesh mesher, availale in pythnocc. You can have a loot at this example: https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_mesh_surfacic.py

The first one (the ventilator) is the result of the mesh performed by the gmesh mesher. It is already available at https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_mesh_gmsh.py

MagaFem, 2018-02-01:
Thanks! Those helped! 

Are there any examples of SMESH meshing a complex volume?

tpaviot, 2018-02-01:
You can try https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_mesh_volumic.py Its meshes a very simple shape. Just replace the line

```
aShape = BRepAlgoAPI_Cut(box, sphere).Shape()
```
with the one used to import the ventilator.stp file, and check the resut

MagaFem, 2018-02-01:
Thanks! 
But that is my problem, that gives an error:
`RuntimeError: StdFail_NotDone
Triangulation not done.`

jf---, 2018-02-02:
@MagaFem , so if a `TopoDS_Shape` is not rendered, you probably need to mesh it explicitly

tpaviot, 2018-02-02:
@MagaFem the error message is quite clear: the triangulation is not performed, but impossible to give you any other information without a bit more material. Please copy/paste your code if you want a more accurate advice.

MagaFem, 2018-02-06:
As suggested above, I changed the file [https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_mesh_volumic.py](https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_mesh_volumic.py) to read in the vent:
`aShape= read_step_file('models/Ventilator.stp')`
I see that the hypotheses 1D hypotheses work fine, but the 2D mesh fails.  I do not know how to properly chose the options to avoid the triangulation error above. Thanks for any help.

MagaFem, 2018-02-07:
After a bit of "playing", I was able to get it to mesh. The following options got 'er done:

```
an1DHypothesis = StdMeshers_Arithmetic1D(0, 0, aMeshGen)
an1DHypothesis.SetLength(2, False)
an1DHypothesis.SetLength(3, True)
an1DAlgo = StdMeshers_Regular_1D(1, 0, aMeshGen)
a2dHypothseis = StdMeshers_TrianglePreference(2, 0, aMeshGen)  # define the boundary
a2dAlgo = StdMeshers_MEFISTO_2D(3, 0, aMeshGen)
a3dHypothesis = StdMeshers_Projection_3D(4, 0, aMeshGen)
```

jf---, 2018-05-31:
@tpaviot can you plz close the issue?

closed 2018-06-13 by MagaFem