tpaviot/pythonocc-core #504: Issues with boolean operations (Cut/Common) on an ellipsoid.

JMlynarz, 2018-02-05:
Dear pythonocc users,

I've encountered an issue when performing some boolean operations. I'm trying to understand what could be the underlying cause of this issue and what I could do to prevent it. 
In some cases, when performing a cut or common boolean operation on a sphere on which I've performed transformations (non-uniform scale and translation), the result of the operation is empty when it shouldn't be. Here's a bit of code that results in such behavior:

from OCC.Display.SimpleGui import init_display

from OCC.gp import gp_Pnt as Pnt
from OCC.gp import gp_XYZ as XYZ
from OCC.gp import gp_Mat as Mat
from OCC.gp import gp_GTrsf as GTrsf
from OCC.BRepPrimAPI import BRepPrimAPI_MakeSphere as makeSphere
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox as makeBox
from OCC.BRepBuilderAPI import BRepBuilderAPI_GTransform as makeGTransform
from OCC.BRepAlgoAPI import BRepAlgoAPI_Common as makeCommon

def main():
    
    display, start_display, add_menu, add_function_to_menu = init_display()
    orig = Pnt(0., 0., 0.)
    shape = makeSphere(orig, 5.).Solid()
    a1 = 171
    a2 = 171
    a3 = 35
    gTrsf = GTrsf(Mat(a1, 0, 0, 0, a2, 0, 0, 0, a3), XYZ(0., 112.2, 0.))
    shape = makeGTransform(shape, gTrsf).Shape()
    box = makeBox(Pnt(-1000, -1000, -1000), Pnt(1000, 112.2, 1000)).Shape()
    display.DisplayShape(makeCommon(box, shape).Shape(), color = "BLACK", transparency = 0.8)

    start_display() 

At first I thought that the issue was only related to the fact that I'm cutting the ellipsoid through its center, but If I change both the value of the translation and the y-end of the box (112.2 -> 110, for instance), the boolean operation succeeds, so it's a bit more arcane, but I cannot find the logic behind it or figure out how to fix it. 

Thanks!
-Jocelyn Mlynarz

tpaviot, 2018-02-06:
I found a tweak: create an original sphere with a greater radius, let's say you multiply by a factor 10., then you have:
```
shape = makeSphere(orig, 50.).Solid()
```
then divide by the same 10 factor the scale_x, scale_y and scale_z parameters:
```
a1 = 17.1
a2 = 17.1
a3 = 3.5
```
Your ellipsoid still has the same size, ok ? But using these parameters, the common boolean operation succeeds. I have absolutely no idea why this fix works, I should dive into the GTransform code.

I uploaded an example to pythonocc-demos (see https://github.com/tpaviot/pythonocc-demos/commit/09fb0b1df0de231c74212e64aea56a515d77e54f), be aware that this example only work with the current development branch (you will note that the .Core package has recently been added).

Please close this issue if ever this also works on your machine.

[image]

JMlynarz, 2018-02-06:
Thank you !

The script indeed works on my machine. However, this was a particular test case I made to debug a more general problem with a quadric generation routine (that builds quadrics based on an arbitrary quadric equation), and it still happens even with smaller scaling coefficients. I've investigated a bit and it seems it's a known issue with OCCT 6.9 (https://tracker.dev.opencascade.org/view.php?id=27390, found other references on the Salome forums), apparently solved in 7.0. I've read somewhere that BRepBuilderAPI_GTransform builds a BSpline surface in this case (non uniform scaling of a sphere), and in the 7.0 changelog it's written that BSpline evaluations/intersections have improved, so I wonder if that's where the problem lies?

closed 2018-03-06 by JMlynarz