tpaviot/pythonocc-core #505: Conda builds are broken.

ikeough, 2018-02-06:
When attempting to follow the instructions [here](https://github.com/tpaviot/pythonocc-core/blob/master/doc/getting_started/build_install.rst) to install pythonocc on Mac OSX, I receive the following:
```
Fetching package metadata .............
Solving package specifications: 

PackageNotFoundError: Packages missing in current channels:
            
  - pythonocc-core -> opencascade_community_edition
  - pythonocc-core -> smesh -> oce 0.16.1

We have searched for the packages in the following channels:
            
  - http://conda.binstar.org/jf/osx-64
  - http://conda.binstar.org/jf/noarch
  - https://repo.continuum.io/pkgs/main/osx-64
  - https://repo.continuum.io/pkgs/main/noarch
  - https://repo.continuum.io/pkgs/free/osx-64
  - https://repo.continuum.io/pkgs/free/noarch
  - https://repo.continuum.io/pkgs/r/osx-64
  - https://repo.continuum.io/pkgs/r/noarch
  - https://repo.continuum.io/pkgs/pro/osx-64
  - https://repo.continuum.io/pkgs/pro/noarch
```

ikeough, 2018-02-06:
This is fixed by https://github.com/tpaviot/pythonocc-core/pull/506. Although, I suggest that the documentation in `/docs` be brought up to date with that in the `README.md`, or removed.

tpaviot, 2018-02-06:
You're right, the doc has to be updated. FYI, I recently moved all the pythonocc doc to another repository http://github.com/tpaviot/pythonocc-documentation . If you want to contribute a patch, please create a PR on that side.