tpaviot/pythonocc-core #509: Examples seem to not run out of the box

wzbernstein, 2018-02-09:
Hi there,
I tried running some of the examples and it seems that the library structure has changed wihout the examples being updated.  Is that right?  For example, the path OCC.core.* is no longer valid apparantly.  Also, I receive other directory errors when finding components in the library, e.g. OCC.extend not found.

Can you comment?

tpaviot, 2018-02-09:
Indeed the structure has changed in the current master, bhe examples have been updated accordingly (see http://github.com/tpaviot/pythonocc-demos). They're now located in the /demos subfolder, which is a git submodule. After you updated the master branch, be sure to:
```
git submodule update --recursive
```

zwn, 2018-05-22:
There is no `OCC.Core` package but the examples you linked to import it (see for example https://github.com/tpaviot/pythonocc-demos/blob/cb06b0e33e3b8d8c483edff9f5e8c9a1f8fed231/examples/core_2d_fillet.py#L18). I have pythonocc-core 0.18 package from pythonocc conda channel.

ImreMD, 2018-05-28:
Same remark, with the test repo after cloning I have to replace all the "OCC.Core.*" with just "OCC.*"  I use a conda python installation and work on Windows 10. Now I am checking two tests errors one probably related to a unpickling issue of box_shape.rep in core_wrapper_features_unittest.py line 319 I think that the problem relates to "\r" ending in the pickle file, the error message is : no module named "copy_reg\r" (see: https://stackoverflow.com/questions/8527241/python-pickle-fix-r-characters-before-loading), the second error is an assertion error related to BRepCheck_ListOfStatus() (line 521) BRepCheck_ListOfStatus() is defined in BRepCheck.py (OCC package) line 1057 but it refers then to the compile module _BRepCheck and it is difficult to have some insight in it, so I don't know for instance what is happening. Just for the record I am a hobbyist (both python and github) .

tpaviot, 2019-01-13:
@ImreMD @zwn pythonocc-demos repository was updated to latest pythonocc-core API changes. It should be released into consistent packages in late February.

closed 2019-01-13 by tpaviot