tpaviot/pythonocc-core #511: Misaligned elements in STEP-import

Schnierle, 2018-02-15:
Good morning pythonocc-community,

I am experiencing problems with the import of some step-Files with pythonocc 0.18.1, Win10.
You can test the behaviour with this File: [turbine_stp.zip](https://github.com/tpaviot/pythonocc-core/files/1726865/turbine_stp.zip)

My test import results:

**Autodesk Inventor 2018 Professional:**
[image]

**FreeCAD-0.16.6712:**
[image]

**Pythonocc 0.18.1 Import with core_load_step_with_colors.py**
[image]

As you can see, there are some misaligned elements in the pythonocc import (highlighted with blue color). I imported the step-File with core_load_step_with_colors.py from the examples-Folder.

Can you reproduce this behaviour? Are there any hints on solving this?

Thanks and best regards
Marc

tpaviot, 2018-02-15:
Thanks for the report, I'll check what's wrong

Schnierle, 2018-02-15:
Thank you very much for your fast reply! 
I have carried out some research and tried to isolate the behavior. It looks like a problem with the drill holes.

[turbine_housing.zip](https://github.com/tpaviot/pythonocc-core/files/1727179/turbine_housing.zip)
**Pythonocc 0.18.1**
[image]

Another example file: [rotary_table.zip](https://github.com/tpaviot/pythonocc-core/files/1727216/rotary_table.zip)

**Pythonocc 0.18.1**
[image]

Interesting point: When I create a cube in Autodesk Inventor and drill a hole into it, we get correct results
 in pythonocc.
See this file: [cube.zip](https://github.com/tpaviot/pythonocc-core/files/1727267/cube.zip)

**Pythonocc 0.18.1**
[image]

Greetings
Marc

Schnierle, 2018-05-31:
Good evening Thomas @tpaviot,

I'm still interested in a solution for the described issue. Did you find anything on this issue or a hint for further research I could do?

Thanks and keep up the great work
Marc

d-zimmermann, 2018-06-16:
Same Issing also here https://github.com/tpaviot/pythonocc-demos/issues/5

tpaviot, 2019-01-13:
Fixed by PR #615 
[image]
[image]

closed 2019-01-13 by tpaviot