tpaviot/pythonocc-core #514: No rule to make target `/usr/X11R6/lib/libGL.dylib', needed by `Darwin/x86_64-Release-64/_Addons.so'.

mjfwest, 2018-02-28:
Hello, 
I followed the instruction for building from source with version #513 on Mac  OS 10.13.3 (17D102) 
I get stuck with the following:
```bash
Scanning dependencies of target _Addons
[ 11%] Building CXX object CMakeFiles/_Addons.dir/Darwin/x86_64-Release-64/AddonsPYTHON_wrap.cxx.o
[ 11%] Building CXX object CMakeFiles/_Addons.dir/src/Addons/Font3d.cpp.o
[ 11%] Building CXX object CMakeFiles/_Addons.dir/src/Addons/TextItem.cpp.o
[ 11%] Building CXX object CMakeFiles/_Addons.dir/src/Addons/LineItem.cpp.o
make[2]: *** No rule to make target `/usr/X11R6/lib/libGL.dylib', needed by `Darwin/x86_64-Release-64/_Addons.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 11%] Building CXX object CMakeFiles/_Addons.dir/src/Addons/TextureItem.cpp.o
make[1]: *** [CMakeFiles/_Addons.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target _StepVisual
[ 11%] Building CXX object CMakeFiles/_StepVisual.dir/Darwin/x86_64-Release-64/StepVisualPYTHON_wrap.cxx.o
Scanning dependencies of target _StepAP214
[ 12%] Building CXX object CMakeFiles/_StepAP214.dir/Darwin/x86_64-Release-64/StepAP214PYTHON_wrap.cxx.o
[ 12%] Linking CXX shared module Darwin/x86_64-Release-64/_StepAP214.so
[ 12%] Built target _StepAP214
[ 12%] Linking CXX shared module Darwin/x86_64-Release-64/_StepVisual.so
[ 12%] Built target _StepVisual
make: *** [all] Error 2
```
Is there a missing build instruction step for Mac OS?  
Will the anaconda packages be updated soon to 0.18.2? I would like to get the Jupyter integration working.

dglb, 2018-06-26:
edit (remove OpenGL)
CMakeFiles/_Addons.dir/link.txt
CMakeFiles/_Addons.dir/build.make

so I can compile PythonOCC

ashenwolf, 2018-07-09:
Same issue on Mac OS 10.13.5. Did anyone manage to solve it?

mjfwest, 2018-07-10:
hello @ashenwolf, have you tried the suggestion by @dglb?  I have not yet.

ashenwolf, 2018-07-10:
@mjfwest not yet as I am not sure what exactly do I need to remove from those files... :(

Robert-M-Muench, 2018-08-03:
Did remove anything with reference to GL from the mentioned files. Now (of course) the linker complains and building stops.

```
[ 12%] Built target GEOMAlgo_swig_compilation
[ 13%] Built target _GEOMAlgo
[ 13%] Built target Addons_swig_compilation
[ 13%] Linking CXX shared module Darwin/x86_64-MinSizeRel-64/_Addons.so
...
Undefined symbols for architecture x86_64:
  "_glBegin", referenced from:
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
  "_glBindTexture", referenced from:
      TextureItem::Init(unsigned char*) in TextureItem.cpp.o
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
  "_glDeleteTextures", referenced from:
      TextureItem::~TextureItem() in TextureItem.cpp.o
  "_glDisable", referenced from:
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
  "_glEnable", referenced from:
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
  "_glEnd", referenced from:
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
  "_glGenTextures", referenced from:
      TextureItem::Init(unsigned char*) in TextureItem.cpp.o
  "_glPixelStorei", referenced from:
      TextureItem::Init(unsigned char*) in TextureItem.cpp.o
  "_glTexCoord2f", referenced from:
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
  "_glTexEnvi", referenced from:
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
  "_glTexImage2D", referenced from:
      TextureItem::Init(unsigned char*) in TextureItem.cpp.o
  "_glTexParameteri", referenced from:
      TextureItem::Init(unsigned char*) in TextureItem.cpp.o
  "_glVertex2f", referenced from:
      TextureItem::RedrawLayerPrs() in TextureItem.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Darwin/x86_64-MinSizeRel-64/_Addons.so] Error 1
make[1]: *** [CMakeFiles/_Addons.dir/all] Error 2
make: *** [all] Error 2
````
Does anyone have an idea how to proceed?

mahaidong, 2018-11-03:
> edit (remove OpenGL)
> CMakeFiles/_Addons.dir/link.txt
> CMakeFiles/_Addons.dir/build.make
> 
> so I can compile PythonOCC

it is not working for me.

tpaviot, 2019-01-13:
There should not be any issue with compiling on osx, see for instance travis builds whatever the xcode version you use (xcode >9). You can try to download install nightly builds for pythonocc-0.18.2 (including Jupyter rendering) at https://anaconda.org/tpaviot/pythonocc-core/files

closed 2019-01-13 by tpaviot