tpaviot/pythonocc-core #515: ACIS-SAT/SAB import-export

mirlip, 2018-03-11:
Hi,

first of all, it is a pleasure to see be able to be able to use your components in python. 
I read here https://www.opencascade.com/content/acis-sat-import-export that opencascade can import ACIS SAT/SAB files, but couldn't find it in the pythonocc-core doc. Am I missing something or is it a closed feature?

Kind regards

jf---, 2018-03-12:
This is a paid for feature that comes as an addition to opencascade

mirlip, 2018-03-14:
ok, thanks for the answer. If I understand correctly, OCCT has code from OCE, which is GPL with the intelectual rights by the people who submitted it? Original OCCT could be dual licensed as it was the sole property of the company managing it. But with it now containing code from OCE (which can only get GPL-compatible licenses) how do they manage to link closed source addons against it? Or are those addon to be paid but still under GPL?
Note that I'm just trying to understand.

tpaviot, 2019-01-11:
The OpenCascade Company develops and sells proprietary components built on top of OCCT. See www.opencascade.com

closed 2019-01-11 by tpaviot

bubuwang007, 2025-03-18:
Can we write a acis or x_t file by pythonocct-core?