tpaviot/pythonocc-core #517: Failed to install for Python 3.6

utkubasyazici, 2018-03-18:
I am trying to install OCC package for both Python 2.7 and 3.6 but I am getting this error for 3.6. 

UnsatisfiableError: The following specifications were found to be in conflict:
  - anaconda -> seaborn==0.8.1=py36h595ecd9_0 -> matplotlib -> freetype[version='>=2.8,<2.9.0a0']
  - pythonocc-core==0.18.1
Use "conda info <package>" to see the dependencies for each package.

Regards,
Utku

tpaviot, 2018-03-19:
This is related to a conflict between the freetype version required/installed by pythonocc (2.7) and the one used by matplotlib (2.8). This is fixed in the current master branch. In the meantime, I suggest you install pythonocc in a dedicated anaconda environment.

UBCdr, 2018-03-19:
There is another issue for windows installation as below. I think packages are not available. 

C:\Users\Utku Basyazici>conda install -c pythonocc pythonocc-core
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - pythonocc-core
  - oce==0.17.3
  - pythonocc-core
  - smesh==6.7.5

Current channels:

  - https://conda.anaconda.org/pythonocc/win-64
  - https://conda.anaconda.org/pythonocc/noarch
  - https://repo.continuum.io/pkgs/main/win-64
  - https://repo.continuum.io/pkgs/main/noarch
  - https://repo.continuum.io/pkgs/free/win-64
  - https://repo.continuum.io/pkgs/free/noarch
  - https://repo.continuum.io/pkgs/r/win-64
  - https://repo.continuum.io/pkgs/r/noarch
  - https://repo.continuum.io/pkgs/pro/win-64
  - https://repo.continuum.io/pkgs/pro/noarch
  - https://repo.continuum.io/pkgs/msys2/win-64
  - https://repo.continuum.io/pkgs/msys2/noarch

thorade, 2018-03-21:
You can add more channels, e.g. the readme mentions some more possibly helpful channels:
https://github.com/tpaviot/pythonocc-core#downloadinstall-binaries

I even added them permanentely, by running:
```
conda config --append channels conda-forge
conda config --append channels DLR-SC
conda config --append channels pythonocc
conda config --append channels oce
```
In this order.
Afterwards, I was able to install or update pythonocc-core with Python 3.6, Anaconda3, Windows10.
This currently gives me pythonocc-core 0.18.1 and oce 0.18.2 and freetype 2.7

tpaviot, 2019-01-12:
Conflicts with freetype versions are fixed in the current master branch.

closed 2019-01-12 by tpaviot