tpaviot/pythonocc-core #522: Extract advanced face names from STEP file

jmauriciou, 2018-03-27:
Hello everybody!

Recently I was able to perform a topology traverse to get primitive coordinates of a STEP geometry. Since I work on a very complex CAD geometry, each face has been named with a corresponding tag. I am able to get all primitive parameters in the geometry. However, I would like to establish the correspondence between face name and primitive parameters.

Consider the following simplified geometry:

[image]

I have assigned a specific tag to the yellow faces. See the STEP file code:

[image]

With the topology traversing I can get the parameters of each plane and cylinder in the entire geometry. Now I want to associate advanced face names with topological entities. In other words, how can I obtain the primitive parameters for these two faces?

I now that this issue requires the comparison of two different objects, i.e. topological entities vs. shape representations. I would appreciate your help. 

Thanks in advance!

Jose

[geometry_names_stp.txt](https://github.com/tpaviot/pythonocc-core/files/1850952/geometry_names_stp.txt)

trelau, 2018-04-12:
If you traverse the topology on the face level, get a "TransferReader" and use it (somewhat) as follows:

```python
reader = STEPControl_Reader()
tr = reader.WS().TransferReader()

# After reading the file and transferring...

item = tr.EntityFromShapeResult(aShape)

# Might need to check for a null handle, but then try:
item.Name().ToCString()
```
That might be enough to get your started. I can perhaps put together a more complete example if needed, that's just a first cut.

icehuli, 2018-04-19:
@jmauriciou 
Hi , I got the same problem. Have you successfully get the names? Could you post the code here. I worked on this for two days and haven't got the right way to do it yet. 
Thanks!

icehuli, 2018-04-19:
@trelau 
Hi, I checked your code. But I always get None form this method EntityFromShapeResult. I don't know why. Could you give a more complete example?
Thanks!

trelau, 2018-04-20:
@icehuli This snippet works below for a similar project of mine:

```python
# Read the file and get the shape
reader = STEPControl_Reader()
tr = reader.WS().TransferReader()
reader.ReadFile('./models/shape_names.step')
reader.TransferRoots()
shape = reader.OneShape()

gui = BasicViewer()

# Explore the faces of the shape (these are known to be named)
exp = TopExp_Explorer(shape, TopAbs_FACE)
while exp.More():
    rgb = None
    s = exp.Current()
    exp.Next()
    item = tr.EntityFromShapeResult(s, 1)
    name = item.Name().ToCString()
    if name:
        print('Found entity: {}'.format(name))
        rgb = (1, 0, 0)
    gui.add(s, rgb)
```
For pythonOCC there might need to some ``.GetObject()`` or ``.GetHandle()`` method calls in there somewhere. The faces I find are shown in red below (CYLINDER_TOP and FACE_UP):
[image]

jmauriciou, 2018-04-20:
Thanks @trelau for your contribution! Just a couple of quick questions. How can I import TransferReader()? I somehow can't make it run.  The same question goes to EntityFromShapeResult(). How can I import them? Thanks!

trelau, 2018-04-20:
@jmauriciou ``TransferReader`` and the ``tr`` variable is just retrieved from the ``STEPControl_Reader`` instance. I haven't had a chance to verify on pythonOCC, so there might be some ``.GetObject()`` or ``GetHandle()`` calls needed since it handles OCC smart pointers differently than pyOCCT. The ``EntityFromShapeResult()`` is just a method of ``tr`` (https://www.opencascade.com/doc/occt-7.2.0/refman/html/class_x_s_control___transfer_reader.html#a8506b9923619cbf3592ef9ec4ee3effe). There is nothing to really import for the two.

There also might be some downcasting required when you get the entity...it might be Handle_Standard_Transient and you have to downcast it to the type that has the Name() method. I have to run now, but can test a pythonocc snippet tonight or this weekend if you still don't have it.

icehuli, 2018-04-22:
@jmauriciou 
Great!! Thanks very much!
This code works for me. 
What I didn't do is 
`item = Handle_StepRepr_RepresentationItem.DownCast(item).GetObject()`

jmauriciou, 2018-04-24:
This is awesome! Thanks!! 

Now one last thing. On the same loop, I would like to get the parameters of the face (location, normal vector) of the face which was identified. I know I can get these parameters using a topology traversing like this:

`
tp = Topo(shape)
     
for face in tp.faces():
            face_adaptor = BRepAdaptor_Surface(face)
            face_type = face_adaptor.GetType()  
            if face_type == GeomAbs_Cylinder:  
                cylinder = face_adaptor.Cylinder()                        
                entity = {}
                entity['type'] = 'cylinder'
                entity['location'] = cylinder.Axis().Location().Coord()
                entity['direction'] = cylinder.Axis().Direction().Coord()
                entity['radius'] = cylinder.Radius()
                entity['coefficients'] = cylinder.Coefficients() 
`

However I would prefer to extract the face name and parameters in the same loop, if possible. 

Greetings!

icehuli, 2018-04-24:
Hi @jmauriciou 

It is basically the same.
```python
from OCC.STEPControl import STEPControl_Reader
from OCC.TopAbs import TopAbs_FACE
from OCC.TopExp import TopExp_Explorer
from OCC.StepRepr import Handle_StepRepr_RepresentationItem
from OCC.BRepAdaptor import BRepAdaptor_Surface
from OCC.GeomAbs import GeomAbs_Cylinder
from OCC.TopoDS import topods

# Read the file and get the shape
reader = STEPControl_Reader()
tr = reader.WS().GetObject().TransferReader().GetObject()
reader.ReadFile('geometry_names.step')
reader.TransferRoots()
shape = reader.OneShape()

# Explore the faces of the shape (these are known to be named)
exp = TopExp_Explorer(shape, TopAbs_FACE)
while exp.More():
    s = exp.Current()
    exp.Next()

    #Converting TopoDS_Shape to TopoDS_Face
    face = topods.Face(s)

    #Working on the geometry
    face_adaptor = BRepAdaptor_Surface(face)
    face_type = face_adaptor.GetType()
    if face_type == GeomAbs_Cylinder:
        cylinder = face_adaptor.Cylinder()
        entity = {}
        entity['type'] = 'cylinder'
        entity['location'] = cylinder.Axis().Location().Coord()
        entity['direction'] = cylinder.Axis().Direction().Coord()
        entity['radius'] = cylinder.Radius()
        entity['coefficients'] = cylinder.Coefficients()
        print('cylinder:', entity)

    #Working on the name
    item = tr.EntityFromShapeResult(s, 1)
    if item.IsNull():
        continue
    item = Handle_StepRepr_RepresentationItem.DownCast(item).GetObject()
    name = item.Name().GetObject().ToCString()
    if name:
        print('Found entity named: {}: {}.'.format(name, s))

```

jmauriciou, 2018-04-24:
Great! Thanks!

closed 2018-04-24 by jmauriciou

jf---, 2018-04-24:
Interesting work. Perhaps worth to add to the pythonocc-example?

youns418, 2019-04-01:
@jmauriciou @trelau @icehuli
Hi, I got a problem. 
I want to make named surface of my 3d shape like the CYLINDER_TOP and FACE_UP.
Could you post the code here?
I worked on this job for a week and haven't got the right way to do it yet.
Thanks!!

youns418, 2019-04-02:
test

youns418, 2019-04-03:
테스트

youns418, 2019-04-03:
test

youns418, 2019-04-04:
test

Dand-D, 2022-04-07:
what if I  want to get the identifier of the face like "#133"and "139" in the picture. what should I do?
[image]

sandeshchand87, 2022-04-27:
Hello All,

from OCC.Core.STEPControl import STEPControl_Reader
from OCC.Core.TopAbs import TopAbs_FACE
from OCC.Core.TopExp import TopExp_Explorer
from OCC.Core.StepRepr import Handle_StepRepr_RepresentationItem

# Read the file and get the shape
file=r"Start.stp"
reader = STEPControl_Reader()
tr = reader.WS().GetObject().TransferReader().GetObject()
reader.ReadFile(file)
reader.TransferRoots()
shape = reader.OneShape()
print('shape',shape)
# Explore the faces of the shape (these are known to be named)
exp = TopExp_Explorer(shape, TopAbs_FACE)
while exp.More():
    s = exp.Current()
    exp.Next()
    item = tr.EntityFromShapeResult(s, 1)
    if item.IsNull():
        continue
    item = Handle_StepRepr_RepresentationItem.DownCast(item).GetObject()
    name = item.Name().GetObject().ToCString()
    if name:
        print('Found entity named: {}: {}.'.format(name, s))
I am getting two errors
mportError                               Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_23672\3372315988.py in <module>
      2 from OCC.Core.TopAbs import TopAbs_FACE
      3 from OCC.Core.TopExp import TopExp_Explorer
----> 4 from OCC.Core.StepRepr import Handle_StepRepr_RepresentationItem
      5 
      6 # Read the file and get the shape

ImportError: cannot import name 'Handle_StepRepr_RepresentationItem' from 'OCC.Core.StepRepr'
stp"
      8 reader = STEPControl_Reader()
----> 9 tr = reader.WS().GetObject().TransferReader().GetObject()
     10 reader.ReadFile(file)
     11 reader.TransferRoots()

AttributeError: 'XSControl_WorkSession' object has no attribute 'GetObject'

Please help me to fix this error.
Many Thanks

flolu, 2022-07-18:
> what if I want to get the identifier of the face like "133"and "139" in the picture. what should I do? [image]

@Dand-D Did you find a way to get the identifier?

im44pos, 2022-09-10:
@trelau @icehuli 
I'm trying to make this work with OCC.Core, and got several errors.
Thus I changed: 
from OCC.Core.StepRepr import Handle_StepRepr_RepresentationItem
tr = reader.WS().GetObject().TransferReader().GetObject()
item = Handle_StepRepr_RepresentationItem.DownCast(item).GetObject()
name = item.Name().GetObject().ToCString()
into 
from OCC.Core.StepRepr import StepRepr_RepresentationItem
tr = reader.WS().GetObject().TransferReader()
item = StepRepr_RepresentationItem.DownCast(item)
name = item.Name().ToCString()
respectively.

Got an AttributeError on: if item.IsNull():
AttributeError: 'Standard_Transient' object has no attribute 'IsNull'

So I commented out:
item = tr.EntityFromShapeResult(s, 1)
    if item.IsNull():
        continue

Nothing is printed, and:
item
<class 'StepRepr_RepresentationItem'>
name
' '

MaFangchang, 2022-10-31:
> @jmauriciou @trelau @icehuli @jf--- Hi, I got a problem. I want to make named surface of my 3d shape like the CYLINDER_TOP and FACE_UP. Could you post the code here? I worked on this job for a week and haven't got the right way to do it yet. Thanks!!

Have you solved this problem now? I am also troubled by this problem. I am now creating a data set of step file, which requires attaching label to each advanced face.

MaFangchang, 2022-10-31:
@jmauriciou Hello, how do you add the names "CYLINDER_TOP" and "FACE_UP" of ADVANCED FACE to the step file. I'm building a dataset containing tens of thousands of step files. I need to name the ADVANCED FACE in the step file, but I don't know how you do this. Please help me, thank you!

jmauriciou, 2022-11-01:
Hello. I don't have any updates other than the contributions on this post.

MaFangchang, 2022-11-11:
> Hello. I don't have any updates other than the contributions on this post.

Thanks for your reply. my problem has been solved.

JorgeFernandes-Git, 2023-06-15:
And it's possible to do the inverse of this?

from let's say this ID:

```
#28=TRIANGULATED_FACE('',#26,10,((-1000.,-0.,-0.)),$,(),((10,9,8),(10,8,7),(10,7,6),(10,6,5),(10,5,4),(10,4,3),(10,3,2),(10,
   2,1)));
```

Get the solids faces edges?