tpaviot/pythonocc-core #530: How to calculate the point of intersection with line and surface?

tnakaicode, 2018-04-19:
I want to calculate the point of intersection with line and surface.

Line is given by a point and direction.
Surface is given by the location and surface data.
If the surface is plane (no curvature), I can calculate the point of intersection by IntAna_IntConicQuad. But, this method can't be applied for curved surface.

In addition, I want to the normal vector at the point of intersection on the surface. 

```import numpy as np
import sys, time, os
import json
import random
from linecache import getline, clearcache
from optparse import OptionParser

from OCC.Display.SimpleGui import init_display
from OCC.gp import gp_Pnt, gp_Ax1, gp_Ax2, gp_Ax3, gp_Vec, gp_Dir
from OCC.gp import gp_Pln, gp_Lin, gp_Translation, gp_Trsf
from OCC.IntAna    import IntAna_IntConicQuad
from OCC.Precision import precision_Angular, precision_Confusion
from OCC.Geom        import Geom_Plane, Geom_Surface, Geom_BSplineSurface
from OCC.Geom        import Geom_Curve, Geom_Line
from OCC.GeomAPI     import GeomAPI_PointsToBSplineSurface, GeomAPI_IntCS
from OCC.GeomAbs     import GeomAbs_C2, GeomAbs_C0, GeomAbs_G1, GeomAbs_G2
from OCC.GeomLProp   import GeomLProp_SurfaceTool
from OCC.Geom2d      import Geom2d_Curve
from OCC.GeomConvert import GeomConvert_ApproxSurface
from OCC.GeomAdaptor import GeomAdaptor_Surface
from OCC.TColgp      import TColgp_Array1OfPnt, TColgp_Array2OfPnt
from OCC.TColStd     import TColStd_Array1OfReal, TColStd_Array1OfInteger
from OCC.TopLoc      import TopLoc_Location
from OCC.TopoDS      import TopoDS_Vertex, TopoDS_Builder, TopoDS_Face
from OCC.STEPControl    import STEPControl_Reader
from OCC.IFSelect       import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCC.BRepBuilderAPI import BRepBuilderAPI_MakeFace
from OCC.BRep           import BRep_Tool
from OCCUtils.Construct import make_plane, make_line, make_edge, make_vertex
from OCCUtils.Construct import dir_to_vec, vec_to_dir
from OCCUtils.Common    import project_point_on_plane
from OCCUtils.Topology  import Topo

def read_step_file(filename):
    """ read the STEP file and returns a compound
    """
    step_reader = STEPControl_Reader()
    status = step_reader.ReadFile(filename)

    if status == IFSelect_RetDone:  # check status
        failsonly = False
        step_reader.PrintCheckLoad(failsonly, IFSelect_ItemsByEntity)
        step_reader.PrintCheckTransfer(failsonly, IFSelect_ItemsByEntity)

        ok = step_reader.TransferRoot()
        _nbs = step_reader.NbShapes()
        aResShape = step_reader.OneShape()
        print (_nbs)
    else:
        print("Error: can't read file.")
        sys.exit(0)
    return aResShape

def rot_axs (axs, ax, deg):
    axs.Rotate(ax, np.deg2rad(deg))

def def_ax (x, y, z, lx=150, ly=200, dx=0, dy=0, dz=0, *args, **kwargs):
    axs = gp_Ax3 (gp_Pnt(x, y, z), gp_Dir(0, 0, 1), gp_Dir(1, 0, 0))
    a_x = gp_Ax1 (gp_Pnt(x, y, z), axs.XDirection())
    a_y = gp_Ax1 (gp_Pnt(x, y, z), axs.YDirection())
    a_z = gp_Ax1 (gp_Pnt(x, y, z), axs.Direction())
    rot_axs (axs, a_x, dx)
    rot_axs (axs, a_y, dy)
    rot_axs (axs, a_z, dz)
    pln = BRepBuilderAPI_MakeFace(
        gp_Pln (axs), -lx/2, lx/2, -ly/2, ly/2
    ).Face()
    return axs, pln

def axs_line (axs):
    p0, dx, dy = axs.Location(), axs.XDirection(), axs.YDirection()
    px = gp_Pnt((gp_Vec(p0.XYZ()) + dir_to_vec(dx)*5 ).XYZ())
    py = gp_Pnt((gp_Vec(p0.XYZ()) + dir_to_vec(dy)*10).XYZ())
    lx = make_line (p0, px)
    ly = make_line (p0, py)
    return lx, ly

def reflect (p0, v0, ax):
    ray = gp_Lin (p0, vec_to_dir(v0))
    intersection = IntAna_IntConicQuad(
        ray, gp_Pln (ax),
        precision_Angular(),
        precision_Confusion()
    )
    p1 = intersection.Point(1)
    vx, vy = gp_Vec(1, 0, 0), gp_Vec(0, 1, 0)
    handle = Geom_Plane(ax)
    handle.D1 (0.5, 0.5, gp_Pnt(), vx, vy)
    vz = vx.Crossed(vy)
    v1 = v0.Mirrored(gp_Ax2(p1, vec_to_dir(vz)))
    return p1, v1

def proj_vec (pln, p, v):
    p0, p1 = p, gp_Pnt((gp_Vec(p.XYZ()) + v).XYZ())
    p2 = project_point_on_plane (pln, p1)
    v2 = gp_Vec (p0, p2)
    return v2

def get_angle (ax, v0):
    p0 = ax.Location()
    vx = dir_to_vec(ax.XDirection())
    vy = dir_to_vec(ax.YDirection())
    vz = dir_to_vec(ax.Direction())
    pl_x = Geom_Plane (p0, ax.XDirection())
    pl_y = Geom_Plane (p0, ax.YDirection())
    v0_x = proj_vec (pl_x, p0, v0)
    v0_y = proj_vec (pl_y, p0, v0)
    dg_x = v0_y.AngleWithRef(vz, vy)*(-1)
    dg_y = v0_x.AngleWithRef(vz, vx)*(-1)
    return dg_x, dg_y

def make_face(px, py, pz):
    nx, ny = px.shape
    print (px.shape, pz.shape)
    pnt_2d = TColgp_Array2OfPnt(1, nx, 1, ny)
    print (pnt_2d.LowerRow(), pnt_2d.UpperRow())
    print (pnt_2d.LowerCol(), pnt_2d.UpperCol())
    for row in range (pnt_2d.LowerRow(), pnt_2d.UpperRow()+1):
        for col in range (pnt_2d.LowerCol(), pnt_2d.UpperCol()+1):
            i,j = row-1, col-1
            pnt = gp_Pnt (px[i, j], py[i, j], pz[i, j])
            pnt_2d.SetValue(row, col, pnt)
            #print (row, col, i, j, px[i, j], py[i, j], pz[i, j])
    curve = GeomAPI_PointsToBSplineSurface(pnt_2d, 3, 8, GeomAbs_G2, 0.001).Surface()
    return curve

if __name__ == '__main__':
    argvs = sys.argv  
    parser = OptionParser()
    parser.add_option("--file", dest="file", default="mou1")
    parser.add_option("--show", dest="show", default=None)
    opt, argc = parser.parse_args(argvs)

    lxy = [float(l) for l in getline (opt.file + ".sfc", 2).split()]
    nxy = [int(n) for n in getline (opt.file + ".sfc", 3).split()]
    px  = np.linspace (lxy[0], lxy[2], nxy[0])
    py  = np.linspace (lxy[1], lxy[3], nxy[1])
    print (px[0], px[-1], px.shape)
    print (py[0], py[-1], py.shape)
    mesh = np.meshgrid (px, py)
    surf = np.loadtxt (opt.file + ".sfc", skiprows=3).T
    spl_geom = make_face (*mesh, surf)
    spl_objt = spl_geom.GetObject()
    spl_face = BRepBuilderAPI_MakeFace(spl_geom, 0, 1, 0, 1, 0.001).Face()
    
    ax0 = gp_Ax3()
    ax1 = gp_Ax3(gp_Pnt(0, 0, 100), gp_Dir(0, 0, 1))
    trf_face = gp_Trsf ()
    trf_face.SetTransformation(ax0, ax1)
    loc_face = TopLoc_Location (trf_face) 
    spl_face.Location(loc_face)
    spl_surf = BRep_Tool.Surface(spl_face)
    print (spl_geom, spl_objt)
    print (spl_face, spl_surf)

    p00, p01 = gp_Pnt(), gp_Pnt() 
    p10, p11 = gp_Pnt(), gp_Pnt() 
    pu, pv   = random.uniform(0,1), random.uniform(0,1)
    pnt = gp_Pnt()
    ray = Geom_Line (gp_Lin (pnt, gp_Dir(0, 0, 1)))
    GeomLProp_SurfaceTool.Value (spl_surf, 0.0, 0.0, p00)
    GeomLProp_SurfaceTool.Value (spl_surf, 0.0, 1.0, p01)
    GeomLProp_SurfaceTool.Value (spl_surf, 1.0, 0.0, p10)
    GeomLProp_SurfaceTool.Value (spl_surf, 1.0, 1.0, p11)
    GeomLProp_SurfaceTool.Value (spl_surf, pu , pv , pnt)
    print (p00)
    print (p01)
    print (p10)
    print (p11)
    print (pu, pv, pnt)
    
    if opt.show:
        display, start_display, add_menu, add_function_to_menu = init_display()
        display.DisplayShape (spl_face)
        display.DisplayShape (gp_Pnt())
        display.DisplayShape (ray)

        display.FitAll()
        start_display()
```

trelau, 2018-04-20:
Try [GeomAPI_IntCS](https://www.opencascade.com/doc/occt-7.2.0/refman/html/class_geom_a_p_i___int_c_s.html). Then, use the method ``GeomAPI_IntCS::Parameters`` to get the (u,v) parameters on the surface. Evaluate the u- and v-direction first derivatives at that point and then take the cross product to get the normal.

tnakaicode, 2018-04-20:
I can solve the problem, and can write the reflection code. Thank you.

```
import numpy as np
import sys, time, os
import random

from OCC.Display.SimpleGui import init_display
from OCC.gp import gp_Pnt, gp_Ax1, gp_Ax2, gp_Ax3, gp_Vec, gp_Dir
from OCC.gp import gp_Pln, gp_Lin, gp_Translation, gp_Trsf
from OCC.Geom           import Geom_Plane, Geom_Surface, Geom_BSplineSurface
from OCC.Geom           import Geom_Curve, Geom_Line
from OCC.GeomAPI        import GeomAPI_PointsToBSplineSurface, GeomAPI_IntCS
from OCC.GeomAbs        import GeomAbs_C0, GeomAbs_C1, GeomAbs_G1, GeomAbs_G2
from OCC.GeomLProp      import GeomLProp_SurfaceTool
from OCC.TColgp         import TColgp_Array1OfPnt, TColgp_Array2OfPnt
from OCC.TColStd        import TColStd_Array1OfReal, TColStd_Array1OfInteger
from OCC.TopLoc         import TopLoc_Location
from OCC.TopoDS         import TopoDS_Vertex, TopoDS_Builder, TopoDS_Face
from OCC.BRepBuilderAPI import BRepBuilderAPI_MakeFace
from OCC.BRep           import BRep_Tool
from OCCUtils.Construct import make_plane, make_line, make_edge, make_vertex
from OCCUtils.Construct import dir_to_vec, vec_to_dir
from OCCUtils.Common    import project_point_on_plane

def make_face(px, py, pz):
    nx, ny = px.shape
    print (px.shape, pz.shape)
    pnt_2d = TColgp_Array2OfPnt(1, nx, 1, ny)
    print (pnt_2d.LowerRow(), pnt_2d.UpperRow())
    print (pnt_2d.LowerCol(), pnt_2d.UpperCol())
    for row in range (pnt_2d.LowerRow(), pnt_2d.UpperRow()+1):
        for col in range (pnt_2d.LowerCol(), pnt_2d.UpperCol()+1):
            i,j = row-1, col-1
            pnt = gp_Pnt (px[i, j], py[i, j], pz[i, j])
            pnt_2d.SetValue(row, col, pnt)
            #print (row, col, i, j, px[i, j], py[i, j], pz[i, j])
    curve = GeomAPI_PointsToBSplineSurface(pnt_2d, 3, 8, GeomAbs_G2, 0.001).Surface()
    return curve

def reflection (h_surf, pnt, vec):
    ray = Geom_Line (gp_Lin (pnt, vec_to_dir(vec.Normalized())))
    uvw = GeomAPI_IntCS (ray.GetHandle(), h_surf).Parameters(1)
    u, v, w = uvw
    p, vx, vy = gp_Pnt(), gp_Vec(), gp_Vec()
    GeomLProp_SurfaceTool.D1 (h_surf, u, v, p, vx, vy)
    vz = vx.Crossed(vy)
    vx.Normalize()
    vy.Normalize()
    vz.Normalize()
    v = v0.Mirrored(gp_Ax2(pnt, vec_to_dir(vz)))
    return p, v
    
if __name__ == '__main__':
    lxy = [-100, -100, 100, 100]
    nxy = [300, 400]
    px  = np.linspace (lxy[0], lxy[2], nxy[0])
    py  = np.linspace (lxy[1], lxy[3], nxy[1])
    print (px[0], px[-1], px.shape)
    print (py[0], py[-1], py.shape)
    mesh = np.meshgrid (px, py)
    surf = mesh[0]**2/500 + mesh[1]**2/600 + mesh[0]*mesh[1]/20000 # arbitrary function or data
    spl_geom = make_face (*mesh, surf)
    spl_objt = spl_geom.GetObject()
    spl_face = BRepBuilderAPI_MakeFace(spl_geom, 0, 1, 0, 1, 0.001).Face()
    
    ax0 = gp_Ax3()
    ax1 = gp_Ax3(gp_Pnt(0, 0, 100), gp_Dir(0, 0, 1))
    trf_face = gp_Trsf ()
    trf_face.SetTransformation(ax1, ax0)
    loc_face = TopLoc_Location (trf_face) 
    spl_face.Location(loc_face)
    spl_surf = BRep_Tool.Surface(spl_face)
    print (spl_geom, spl_objt)
    print (spl_face, spl_surf)

    p0, v0 = gp_Pnt(), gp_Vec(0, -0.1, 1.0).Normalized()
    p1, v1 = reflection (spl_surf, p0, v0)
    p2 = gp_Pnt((gp_Vec(p1.XYZ()) + v1*50).XYZ())
    ray01 = make_edge (p0, p1)
    ray12 = make_edge (p1, p2)
    
    display, start_display, add_menu, add_function_to_menu = init_display()
    display.DisplayShape (spl_face)
    display.DisplayVector (v0*10, p0)
    display.DisplayVector (v1*10, p1)
    display.DisplayShape (ray01)
    display.DisplayShape (ray12)

    display.FitAll()
    start_display()

```

closed 2018-04-20 by tnakaicode