tpaviot/pythonocc-core #531: Installation issue with Anaconda (2&3)

guillaumesimler, 2018-04-22:
Cher Thomas,

I face the following challenge when installing the package with Anaconda. 

After the installation of your conda package, I will get the following error.
```
(vel_proto) C:\Users\Guillaume\Desktop\Velchanos Front End>deactivate
The system cannot find the path specified.

(vel_proto) C:\Users\Guillaume\Desktop\Velchanos Front End>
```
Anything works except **internal conda use**:

- you can run python, jupyter, etc...
- you can use conda outside the environment and update/add additional libraries

But you can deactivate, add libraries within the conda environment. 
  
[vel_proto.zip](https://github.com/tpaviot/pythonocc-core/files/1935924/vel_proto.zip)

Could you please help me ?

Merci beaucoup et bonne continuation

Guillaume

glinskyc, 2018-09-07:
I too encountered something similar.  For me this is the issue:

https://github.com/ContinuumIO/anaconda-issues/issues/7581

The issue above is unresolved at time of writing this, and nobody is working on it from what I can tell.

Unfortunately, I still do not have a true solution.  The only way I'm able to use conda within the active conda environment is to run all terminal commands from this dir:

`C:\Users\<UserName>\Miniconda3\Scripts`

The root of the issue seems to be that python itself is installed as part of the pythonocc installation process, which causes confusion for conda.  I suppose my question for Dr. Paviot (@tpaviot) is why does python itself need to be installed as part of the pythonocc installation, rather than leveraging a user's existing python installation?

tpaviot, 2019-01-12:
I confirm the bug for pythonocc-core 0.18.1. Fixed in the current master branch. Current packages for nightly builds (see https://anaconda.org/tpaviot/pythonocc-core) work as expected, ie you can activate/deactivate any environment.

closed 2019-01-12 by tpaviot