tpaviot/pythonocc-core #532: off-screen rendering on Mac

mounte, 2018-05-01:
So I am pretty new to the mac-scene and are facing the problem that when the off-screen renderer is initialized I get the following errors:

RuntimeError: Aspect_WindowDefinitionError
Cocoa application should be instantiated before window

What I am trying to do is build a simple rest-api that will generate a rendering of a part and return it as a jpg. Seems that a Cocoa app need to be created. Does anyone have some sample code that get the off-screen renderer to work under Mac OSX?

jf---, 2018-05-01:
IIRC only `linux` has a headless capacity ( this is a feature of the OpenGL driver that osx doesnt support )

Have you looked into the `Three.js` rendering facilities? 
Might be relevant for a REST api

tpaviot, 2019-01-13:
Offscreen rendering needs a graphic system (X.org or xvfb on Linux, Cocoa on OSX). Even on a server side app based on pythonocc, you need opengl to be installed on the server if you need to generate thumbnails. But you don't need any GUI.

closed 2019-01-13 by tpaviot

pridi, 2021-03-09:
@tpaviot I'm having the same issue as @mounte has described. What do you mean for "graphic system"? How could I generate thumbnails in macOS? Thank you very much