tpaviot/pythonocc-core #537: Wrap Transfer module

trelau, 2018-05-20:
I was trying to write a script for https://github.com/tpaviot/pythonocc-core/issues/482 but it seems the `Transfer` module is not wrapped and my process needed `Transfer_FinderProcess` to set the STEP entity names. OCC 6.9.1 doc is here: https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_transfer___finder_process.html

Here is the start of the script that demonstrates the issue:

```python
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.STEPControl import STEPControl_Writer

box = BRepPrimAPI_MakeBox(10,10,10).Solid()

writer = STEPControl_Writer()
fp = writer.WS().GetObject().TransferWriter().GetObject().FinderProcess()

print(fp)

fp.GetObject()
```

The output is:
```bash
<Swig Object of type 'Handle_Transfer_FinderProcess *' at 0x000001E5A839F840>
swig/python detected a memory leak of type 'Handle_Transfer_FinderProcess *', no destructor found.
```

hducg, 2018-09-28:
Hi, I have the same problem. Did you find a solution?

trelau, 2018-09-29:
@hducg the solution for me was this: https://github.com/LaughlinResearch/AFEM/blob/master/examples/exchange_step_write.py

This method in Python shows the alternative pythonocc tools that could be used perhaps (`STEPConstruct`): https://github.com/LaughlinResearch/AFEM/blob/master/afem/exchange/step.py#L102-L118

tpaviot, 2018-11-12:
Fixed by commit ca93e887bc2f1587798d65181140b679aa57c900

closed 2018-11-12 by tpaviot