tpaviot/pythonocc-core #538: Remove dependency on pyqt from conda package

zwn, 2018-05-25:
Could the dependency on `pyqt` be removed from the conda package? AFAIK it is not really required to use `pythonocc-core` and it has the unfortunate effect of switching everything from LGPL to GPL.

jf---, 2018-05-28:
>  it has the unfortunate effect of switching everything from LGPL to GPL

the `pyqt` license has no effect on `pythonocc`'s license. if this is an issue for you, why not use `Qt for Python` ( formerly known as pyside2, but adapted by the Qt company )

zwn, 2018-05-29:
I'd like to use something else but when I install `pythonocc-core` conda package, it drags in `pyqt` as an dependency. Please see:
https://github.com/tpaviot/pythonocc-core/blob/5d5038312f0fddd1c64a63f4fbdb59eaab37d708/ci/conda/meta.yaml#L33
I am just proposing to have this line deleted as AFAIU `pythonocc-core` does not really depend on `pyqt` (it can use it, if it is available, but it works without it).

tpaviot, 2018-11-15:
@zwn You're right, ```pythonocc-core``` can be used in a GUIless purpose. Adding pyqt as a dependency was a way to ensure that examples will actually run without any additional user action. But I agree it could be a good thing that pythonocc-core conda package is made independent from any GUI pakcage. pyqt could have the same status as pythreejs in the current master : something that the user installs if he wants to benefit the feature. @jf--- your opinion ?

tpaviot, 2018-11-15:
Fixed in branch https://github.com/tpaviot/pythonocc-core/tree/review/remove-pyqt-as-dep

closed 2019-01-11 by tpaviot