tpaviot/pythonocc-core #540: How should I cite python-occ?

cristobaltapia, 2018-06-08:
Hi @tpaviot ! 
Thanks a lot for this nice python package! 
I am using Python-occ for my research and would like to cite it. How should I do it?

jf---, 2018-06-11:
Perhaps following this paper; [An Advanced STEP-NC Platform for Additive Manufacturing](https://www.researchgate.net/publication/319509472_An_Advanced_STEP-NC_Platform_for_Additive_Manufacturing), where `pythonocc` is cited as:

T. Paviot, pythonOCC, 3D CAD/CAE/PLM development framework for the Python 
programming language, PythonOCC – 3D CAD Python.
http://www.pythonocc.org/ (accessed January 19, 2017). 

PS: perhaps we could feature your work on the `pythonocc` site? we've has a nr of such posts, they're useful and exciting contributions ;)

tpaviot, 2018-06-12:
@cristobaltapia You can use the suggestion from @jf--- 

@aothms I started looking at the JOSS a couple of months ago, but then I was commited in another project and delayed the submission. I do think it's the more efficient/practical/fastest way to have a peer reviewed paper to be used as a scientific reference.

aothms, 2018-06-13:
> I do think it's the more efficient/practical/fastest way to have a peer reviewed paper to be used as a scientific reference.

I think so too, I'm also still planning to write something. A 'conventional' journal article might get a bigger audience though. If you don't want to contribute to the existing monopolies and don't mind focussing on a construction audience, you can also consider itcon.org