tpaviot/pythonocc-core #541: Conda installation under Python 3.6 on Mac

ghareth, 2018-06-12:
Also reported on [https://github.com/tpaviot/oce/issues/692]( https://github.com/tpaviot/oce/issues/692). Moved issue as suggested by @aothms 

I have been running pythonocc successfully installed using conda under Python 2.7 within both my Windows(32-bit Python) and MacOS (64-bit Python) environments.

Upgrading my pythonocc installations to Python 3.6, pythonocc has successfully installed in Windows (using 32-bit Python) while the MacOs installation suffers from the following error:

```
import OCC.gp 

Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```

@aothms commented ...

I also ran into this publishing builds for my own project. What I ended up doing is not to link to the python dylib at compile time and allow the linker to ignore the missing symbols. It's been years since I looked into this, but I remember this to be somewhat common practice on OS X.

Set flags:[ https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.6.0/conda/build.sh#L10]( https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.6.0/conda/build.sh#L10)
Link conditionally: [https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.6.0/src/ifcwrap/CMakeLists.txt#L59](https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.6.0/src/ifcwrap/CMakeLists.txt#L59)

@danZib posted workaround ... 
The issue was fixed for me after switching miniconda3 for miniconda2

rainman110, 2018-06-16:
The reason is, that pythonocc core may not be linked anymore to libpython. Please have a look here:https://groups.google.com/a/continuum.io/forum/m/#!topic/anaconda/057P4uNWyCU

tpaviot, 2019-01-13:
Fixed by PR #613 thanks @rainman110 for the patch

closed 2019-01-13 by tpaviot