tpaviot/pythonocc-core #545: Problem with BRepBuilderAPI_MakeFace() function

duncanusn, 2018-07-31:
Dear sir

I try to create shell square with a small hollow square inside (pic. looks like below)
[image]

I believe there' s some misuse for my script and I came up with an error "RuntimeError: StdFail_NotDone"...
It' s a little bit hard for new pythonocc user to debug the error message.

I would be grateful if you give me some advice.
Here's my code. (Basically I just create two wire squares and try to make it into face)
```test.py
#-*- coding: utf-8 -*-
__author__ = 'steve'
__version__= 'Anaconda3-4.4.0-Windows-x86_64 with pythonOCC 0.18.1'

# general library
import sys
import os
import itertools

# pythonOCC library
#
from OCC.gp import gp_Pnt
from OCC.GC import GC_MakeSegment
from OCC.BRepBuilderAPI import \
  BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace, BRepBuilderAPI_MakeWire,\
  BRepBuilderAPI_MakeShell, BRepBuilderAPI_MakeSolid, BRepBuilderAPI_Sewing
from OCC.BRep import BRep_Builder
from OCC.TopoDS import TopoDS_Shell, TopoDS_Solid, TopoDS_Wire, TopoDS_Edge
from OCC import StlAPI
from OCC.Display.SimpleGui import *

# numpy
#
import numpy as np

builder = BRep_Builder()
shell = TopoDS_Shell()
builder.MakeShell(shell)

pts_array_1 = [[0,0,0],
             [0,1,0],
             [1,1,0],
             [1,0,0],
             [0,0,0]]
pts_array_2 =[[0.25,0.25,0],
             [0.25,0.75,0],
             [0.75,0.75,0],
             [0.75,0.25,0],
             [0.25,0.25,0]]

# change numpy real type to standard float
#
edges = []
pts_array_1 = [gp_Pnt(float(p[0]), float(p[1]), float(p[2])) for p in pts_array_1]
length = len(pts_array_1)
for i in range(length-1):
    cur = i
    nxt = i+1
    edges.append(BRepBuilderAPI_MakeEdge(pts_array_1[cur], pts_array_1[nxt]))

pts_array_2 = [gp_Pnt(float(p[0]), float(p[1]), float(p[2])) for p in pts_array_2]
length = len(pts_array_2)
for i in range(length-1):
    cur = i
    nxt = i+1
    edges.append(BRepBuilderAPI_MakeEdge(pts_array_2[cur], pts_array_2[nxt]))

wire = BRepBuilderAPI_MakeWire()
for edge in edges:

    wire.Add(edge.Edge())
    # builder.Add(wires, edge.Shape())
oFace = BRepBuilderAPI_MakeFace(wire.Wire())
builder.Add(shell, oFace.Shape())

display, start_display, add_menu, add_function_to_menu = init_display()
display.DisplayShape(shell, update=True)
start_display()

```
Thanks,

Steve

jf---, 2018-07-31:
@duncanusn you need to take care of the winding of the edges / wires; that is the orientation of the smaller wire should be the inverse of the wire bounding the face

duncanusn, 2018-08-01:
Dear @jf--- 
Thank you for your advice. 
Although I changed the orientation for the smaller wire from "CW" to "CCW", it still didn't work.
Any suggestions? (modified code as below:)
```test.py
#-*- coding: utf-8 -*-
__author__ = 'steve'
__version__= 'Anaconda3-4.4.0-Windows-x86_64 with pythonOCC 0.18.1'

# general library
import sys
import os
import itertools

# pythonOCC library
#
from OCC.gp import gp_Pnt
from OCC.GC import GC_MakeSegment
from OCC.BRepBuilderAPI import \
  BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace, BRepBuilderAPI_MakeWire,\
  BRepBuilderAPI_MakeShell, BRepBuilderAPI_MakeSolid, BRepBuilderAPI_Sewing
from OCC.BRep import BRep_Builder
from OCC.TopoDS import TopoDS_Shell, TopoDS_Solid, TopoDS_Wire, TopoDS_Edge
from OCC import StlAPI
from OCC.Display.SimpleGui import *

# numpy
#
import numpy as np

builder = BRep_Builder()
shell = TopoDS_Shell()
builder.MakeShell(shell)

pts_array_1 = [[0,0,0],
             [0,1,0],
             [1,1,0],
             [1,0,0],
             [0,0,0]]
pts_array_2 =[[0.25,0.25,0],
             [0.25,0.75,0],
             [0.75,0.75,0],
             [0.75,0.25,0],
             [0.25,0.25,0]]

# change numpy real type to standard float
#
edges = []
pts_array_1 = [gp_Pnt(float(p[0]), float(p[1]), float(p[2])) for p in pts_array_1]
length = len(pts_array_1)
for i in range(length-1):
    cur = i
    nxt = i+1
    edges.append(BRepBuilderAPI_MakeEdge(pts_array_1[cur], pts_array_1[nxt]))

pts_array_2 = [gp_Pnt(float(p[0]), float(p[1]), float(p[2])) for p in pts_array_2]
length = len(pts_array_2)
for i in range(length-1, 0, -1):
    cur = i
    nxt = i-1
    edges.append(BRepBuilderAPI_MakeEdge(pts_array_2[cur], pts_array_2[nxt]))

wire = BRepBuilderAPI_MakeWire()
for edge in edges:

    wire.Add(edge.Edge())
    # builder.Add(wires, edge.Shape())
oFace = BRepBuilderAPI_MakeFace(wire.Wire())
builder.Add(shell, oFace.Shape())

display, start_display, add_menu, add_function_to_menu = init_display()
display.DisplayShape(shell, update=True)
start_display()
```

rainman110, 2018-08-20:
I am pretty sure, that you cannot create a shell from only wires or faces that do not create a closed shell. What you really want is a TopoDS_Face with an internal trimming curve, that removes the internal part of the face.