tpaviot/pythonocc-core #546: No test / example working... (ubuntu)

Robert-M-Muench, 2018-08-03:
I compiled and installed everything necessary and now get:

````
$ python core_2d_fillet.py 
Traceback (most recent call last):
  File "core_2d_fillet.py", line 18, in <module>
    from OCC.gp import gp_Pnt,gp_Dir,gp_Pln
  File "/usr/lib/python2.7/dist-packages/OCC/gp.py", line 24, in <module>
    _gp = swig_import_helper()
  File "/usr/lib/python2.7/dist-packages/OCC/gp.py", line 23, in swig_import_helper
    return importlib.import_module('_gp')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _gp
````

And I don't have a clue what to do now...

jf---, 2018-08-31:
Why not install via conda? This will get you up and running in no time.

AtomicFireMind, 2019-09-30:
I encounter the same problem on Ubuntu18.04 with github repo (pythonocc-core 0.18.3 ? and oce 0.18.3? on Sep.29). In the python interpreter command, import OCC without no error, but facing issue, no module named "_Standard", "_Aspect"..... plus, the compling process going through with no error.
And I ever try "conda install -c tpaviot -c oce -c pythonocc pythonocc-core=0.18.2 python=3.6" before I try to compile by myself, but never successed. 
Could you tell what maybe the problem about my complie ?

AtomicFireMind, 2019-09-30:
I fixed conda install on my computer following others comments, with "condarc" config as bellow:
auto_activate_base: False
channels:
  - pythonocc
  - tpaviot
  - oce
  - 3dhubs
  - dlr-sc
  - conda-forge
  - defaults
Finally it run the test success. **But** I still feel doubt about the self compiling, what is the problem?