tpaviot/pythonocc-core #551: Off-Screen Render in Windows 10 64-Bit : RuntimeError

Vishnu-C, 2018-08-19:
Following Error appears while using Off-Screen Render in Windows 10 64-Bit,

Error while executing the demo file core_offscreen_rendering.py

```
C:\Python27\lib\site-packages\OCC\Quantity.py:3: DeprecationWarning: OCC.Quantity is deprecated since pythonocc-0.18.2. Use OCC.Core.Quantity                                                                        
warnings.warn("OCC.Quantity is deprecated since pythonocc-0.18.2. Use OCC.Core.Quantity", DeprecationWarning)                          
###### 3D rendering pipe initialisation #####                                                                                              
Display3d class initialization starting ...                                                                                                
Aspect_DisplayConnection created.                                                                                                          
Graphic_Driver created.                                                                                                                    
V3d_Viewer created.                                                                                                                        
AIS_InteractiveContext created.                                                                                                            
V3d_View created                                                                                                                           
Standard GLSL programs are not found in: C:/Miniconda-x64/conda-bld/oce_1502607315844/_h_env/Library/share/oce/src/Shaders                 
Traceback (most recent call last):                                                                                                         
   File "core_offscreen_rendering.py", line 24, in <module>                                                                                
      offscreen_renderer.Create()                                                                                                         
   File "C:\Python27\lib\site-packages\OCC\Display\OCCViewer.py", line 178, in Create                                                      
       self.InitOffscreen(640, 480)                                                                                                       
RuntimeError: Standard_Failure                                                                                                             CSF_ShadersDirectory or CASROOT is set incorrectly 
```

Is it anything to deal with my graphics card?
Thanks

Vishnu-C, 2018-08-20:
Found the cause,
CASROOT is not set as below

https://github.com/tpaviot/pythonocc-core/blob/5d5038312f0fddd1c64a63f4fbdb59eaab37d708/src/Display/OCCViewer.py#L67

**Solution**
Create a Environment variable CASROOT and set to "full_path_**/Library/share/oce**"

Thanks

smhty, 2018-08-30:
Hi @Vishnu-C, 
I have similar issue.
I applied your method, but that didn't work for me.
Any suggestion???

closed 2018-12-28 by Vishnu-C

weiweiisok, 2023-11-27:
> Hi @Vishnu-C, I have similar issue. I applied your method, but that didn't work for me. Any suggestion???

Hi, I also have this problem. Did you have solve this problem?

Vishnu-C, 2023-11-27:
@smhty @weiweiisok 
I fixed it by creating a Environment variable CASROOT. Take a look above for my [comment](https://github.com/tpaviot/pythonocc-core/issues/551#issuecomment-414378610).