tpaviot/pythonocc-core #554: How to split a TopoDS_Shape ?

tnakaicode, 2018-09-06:
I want to split the box by plane.

I can make the intersection between box and plane used by BRepAlgoAPI_Section.
But, I cannot split the box. I used BRepFeat_SplitShape.

I got following error.

```
TypeError: in method 'BRepFeat_SplitShape_Add', argument 2 of type 'TopoDS_Edge const &'
```
I want to know how to use BRepFeat_SplitShape_Add in my code.

```
import numpy as np
import matplotlib.pyplot as plt 

from OCC.Display.SimpleGui import init_display
from OCC.gp import gp_Pnt, gp_Vec, gp_Dir, gp_Ax1, gp_Ax2, gp_Ax3, gp_Trsf
from OCC.gp import gp_Lin, gp_Pln, gp_Elips
from OCC.TopoDS import TopoDS_Shape 
from OCC.TopoDS import TopoDS_Wire, TopoDS_Edge, TopoDS_Face 
from OCC.TopExp import TopExp_Explorer 
from OCC.TopAbs import TopAbs_EDGE, TopAbs_FACE, TopAbs_REVERSED
from OCC.BRepFeat    import BRepFeat_SplitShape
from OCC.BRepAlgoAPI import BRepAlgoAPI_Section
from OCC.BRepBuilderAPI import BRepBuilderAPI_MakeWire
from OCCUtils.Construct import dir_to_vec, vec_to_dir
from OCCUtils.Construct import make_box, make_plane, make_wire, make_face
from OCCUtils.Topology  import Topo, dumpTopology

if __name__ == "__main__":
    display, start_display, add_menu, add_function_to_menu = init_display()

    base = make_box (100, 100, 100)
    p, v = gp_Pnt(50, 50, 50), gp_Vec(1, 1, 1)
    sect = BRepAlgoAPI_Section (base, gp_Pln(p, vec_to_dir(v)))
    sect.Build()
    edge = sect.Shape()

    display.DisplayShape (gp_Pnt())
    display.DisplayShape (base)
    display.DisplayShape (edge)

    splt = BRepFeat_SplitShape (base)
    wire = []
    Ex = TopExp_Explorer (edge, TopAbs_EDGE)
    while Ex.More():
        print (Ex.Current())
        Sx = TopExp_Explorer (base, TopAbs_FACE)
        while Sx.More():
            if sect.HasAncestorFaceOn1(Ex.Current(), Sx.Current()):
                print (Ex.Current(), Sx.Current())
                splt.Add (Ex.Current(), Sx.Current())
            Sx.Next()
        Ex.Next()
    
    display.FitAll()
    start_display ()
```

tnakaicode, 2018-09-06:
Please refer this page.
https://www.opencascade.com/content/how-split-topodsshape

jf---, 2018-09-11:
Actually this line in the topology iteration loop gave a pretty precise hint:

>   print(Ex.Current(), Sx.Current())
>  class<'TopoDS_Shape'; Type:Edge; id:1814432081>
>  class<'TopoDS_Shape'; Type:Edge; id:1814432081> class<'TopoDS_Shape'; Type:Face; id:1819061553>

And indeed, the `BRepAlgoAPI_Section.HasAncestorFaceOn1 ` method takes a `TopoDS_Shape , TopoDS_Shape`, respectively an edge and face as arguments.

The `BRepFeat_SplitShape.Add` doesn't take a `TopoDS_Shape` argument however, so that's the culprit here.

```python
    def Add(self, *args) -> "void":
        """
        * Adds the wire <W> on the face <F>. Raises NoSuchObject if <F> does not belong to the original shape.

        :param W:
        :type W: TopoDS_Wire &
        :param F:
        :type F: TopoDS_Face &
        :rtype: None

        * Adds the edge <E> on the face <F>.

        :param E:
        :type E: TopoDS_Edge &
        :param F:
        :type F: TopoDS_Face &
        :rtype: None

        * Adds the compound <Comp> on the face <F>. The compound <Comp> must consist of edges lying on the face <F>. If edges are geometrically connected, they must be connected topologically, i.e. they must share common vertices. //! Raises NoSuchObject if <F> does not belong to the original shape.

        :param Comp:
        :type Comp: TopoDS_Compound &
        :param F:
        :type F: TopoDS_Face &
        :rtype: None

        * Adds the edge <E> on the existing edge <EOn>.

        :param E:
        :type E: TopoDS_Edge &
        :param EOn:
        :type EOn: TopoDS_Edge &
        :rtype: None

        """
```

jf---, 2018-09-11:
The new [`volume maker` API](https://www.opencascade.org/doc/overview/html/occt_user_guides__boolean_operations.html#occt_algorithms_10b) is a good fit for splitting the volume, here's an example

```python
from OCC.Core.BOPAlgo import BOPAlgo_MakerVolume
from OCC.Core.BRep import BRep_Builder
from OCC.Core.TopTools import TopTools_ListOfShape
from OCC.Core.TopoDS import TopoDS_Compound
from OCC.Core.gp import gp_Pln
from OCC.Core.gp import gp_Pnt, gp_Vec
from OCC.Display.SimpleGui import init_display
from OCCUtils import Topo
from OCCUtils.Construct import make_box, make_face
from OCCUtils.Construct import vec_to_dir

def make_compound(topology_to_compound):
    """

    :type topology_to_compound: iterable
    """
    compound = TopoDS_Compound()
    builder = BRep_Builder()
    builder.MakeCompound(compound)
    for shp in topology_to_compound:
        builder.Add(compound, shp)

    return compound

def common(shapes):
    mv = BOPAlgo_MakerVolume()
    ls = TopTools_ListOfShape()
    for i in shapes:
        ls.Append(i)
    mv.SetArguments(ls)
    mv.Perform()

    print("error status: {}".format(mv.ErrorStatus()))

    return mv.Shape()

def main():
    display, start_display, add_menu, add_function_to_menu = init_display()

    box = make_box(100, 100, 100)

    p, v = gp_Pnt(50, 50, 50), gp_Vec(0, 0, -1)
    pln = gp_Pln(p, vec_to_dir(v))
    fc = make_face(pln, -1000, 1000, -1000, 1000)  # limited, not infinite plane

    display.DisplayShape(fc, transparency=0.5)
    common_shape = common([box, fc])

    solids = Topo(common_shape).solids()
    display.DisplayShape(next(solids), transparency=0.5)
    display.DisplayColoredShape(next(solids), "BLACK")

    display.FitAll()
    start_display()

if __name__ == "__main__":
    main()
```

[image]

tnakaicode, 2018-09-15:
This is good idea. I tried to write following code in reference your code.

```
from OCC.Display.SimpleGui import init_display
from OCC.gp import gp_Pln
from OCC.gp import gp_Pnt, gp_Vec
from OCC.TopoDS  import TopoDS_Compound
from OCC.BOPAlgo import BOPAlgo_MakerVolume, BOPAlgo_Builder
from OCC.BRep    import BRep_Builder
from OCCUtils.Topology  import Topo
from OCCUtils.Construct import make_box, make_face
from OCCUtils.Construct import vec_to_dir

if __name__ == "__main__":
    display, start_display, add_menu, add_function_to_menu = init_display()

    box = make_box(100, 100, 100)

    p1, v1 = gp_Pnt(50, 50, 50), gp_Vec(0, 0, -1)
    fc1 = make_face(gp_Pln(p1, vec_to_dir(v1)), -1000, 1000, -1000, 1000)  # limited, not infinite plane
    
    p2, v2 = gp_Pnt(50, 50, 50), gp_Vec(0, 1, -1)
    fc2 = make_face(gp_Pln(p2, vec_to_dir(v2)), -1000, 1000, -1000, 1000)  # limited, not infinite plane

    p3, v3 = gp_Pnt(50, 50, 25), gp_Vec(1, 1, -1)
    fc3 = make_face(gp_Pln(p3, vec_to_dir(v3)), -1000, 1000, -1000, 1000)  # limited, not infinite plane

    bo = BOPAlgo_Builder()
    bo.AddArgument (box)
    bo.AddArgument (fc1)
    bo.AddArgument (fc2)
    bo.AddArgument (fc3)

    bo.Perform()
    print("error status: {}".format(bo.ErrorStatus()))

    colos = ["BLUE", "RED", "GREEN", "YELLOW", "BLACK", "WHITE", "BLUE", "RED"]

    top = Topo (bo.Shape())
    for i, sol in enumerate(top.solids()):
        display.DisplayShape(sol, color=colos[i], transparency=0.5)
    
    display.FitAll()
    start_display()
```

And, I got this solid.

[image]

The shape is splited in two shapes by one plane. I want to get just one shape "under the plane".
I define "under the plane" as the negative area of z-axis used to define the plane.

tpaviot, 2018-12-07:
Try the GEOMAlgo_Splitter class see https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_splitter.py

closed 2019-01-11 by tpaviot