tpaviot/pythonocc-core #555: Question about creating a new 3d coordinate system

cityofstaar, 2018-09-10:
Hello guys! 

I'm a beginner to pythonOCC. 

I have a question to create a new coordinate system.

My purpose is to create a point p2 which has (1,1,1) on a new 3d coordinate system named a1.

I think I completed to create a1 correctly but I don't have any idea to create p2.

p2=gp_Pnt(1,1,1) stands for creating a point on a global coordinate system right?

my code is below.

def axis():
    p = gp_Pnt(1., 0., 0.)
    d = gp_Dir(1., 0., 0.)
    a1 = gp_Ax2(p, d)
    
    p2=????
    display.DisplayShape(p2, update=True)
    display.DisplayMessage(p2, "P2")

 axis()
Please give me direction to solve it. thank you very much.

tnakaicode, 2018-09-10:
Set p2=a1.Location().

Show the API reference (https://cdn.rawgit.com/tpaviot/pythonocc-core/804f7f3/doc/apidoc/0.18.1/OCC.gp.html?highlight=gp_ax2#OCC.gp.gp_Ax2) or source code of PyhtonOCC.

cityofstaar, 2018-09-11:
@tnakaicode thank you for your comment. 
But I think a1.location() just create an origin point of a1 coordinate system.. Am I wrong?

jf---, 2018-09-11:
> p2=gp_Pnt(1,1,1) stands for creating a point on a global coordinate system right?

Look into the `.Transform` method to transform from one coordinate system to another

tnakaicode, 2018-09-11:
I misunderstood.

I often use following. 

```
p1 = gp_Pnt(1., 0., 0.)
a1 = gp_Ax2(p1, gp_Dir(1, 0, 0), gp_Dir(0, 0,1))

p2 = gp_Pnt (1, 1, 1)
a2 = gp_Ax3 (p2, gp_Dir(0, 0, 1))

trf = gp_Trsf()
trf.SetTransformation (gp_Ax3(a1), gp_Ax3())
a2.Transform (trf)
loc_p2 = a2.Location()
```

cityofstaar, 2018-09-12:
@tnakaicode thank you! your comment helped me a lot.

closed 2019-01-12 by tpaviot