tpaviot/pythonocc-core #557: Error: No module named 'core_topology_traverse' in pythonocc

akhilesh810, 2018-09-27:
import random
import os
import os.path
import sys

from OCC.STEPControl import STEPControl_Reader
from OCC.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCC.Display.SimpleGui import init_display
from OCC.BRep import BRep_Tool_Surface
from OCC.GeomAbs import GeomAbs_Plane, GeomAbs_Cylinder
from OCC.TopoDS import topods_Face
from OCC.BRepAdaptor import BRepAdaptor_Surface
from core_topology_traverse import Topo

ImportError                               Traceback (most recent call last)
<ipython-input-9-ee78c8a01adf> in <module>()
     11 from OCC.TopoDS import topods_Face
     12 from OCC.BRepAdaptor import BRepAdaptor_Surface
---> 13 from core_topology_traverse import Topo
ImportError:  No module named 'core_topology_traverse'

I am getting an error " No module named 'core_topology_traverse.

Can anyone help what should I do?
Do I need to install any packages? If yes which package should I install?
Thanks in advance

tpaviot, 2019-01-11:
API changed. Use the following line:
```from OCC.Extend.Topologyutils import TopologyExplorer```

closed 2019-01-11 by tpaviot

francisbitontistudio, 2019-02-20:
Still getting an error 
ImportError: No module named Extend.Topologyutils

francisbitontistudio, 2019-02-21:
Got it to work I had to use

from OCC.TopExp import TopExp_Explorer

The issue I am having now is that now I am trying to get information from a BSplineSurface

    elif surf_type == GeomAbs_BSplineSurface:
        print("--> BSplineSurface")
        gp_bsrf = surf.BSplineSurface()

Every time this function runs it crashes

AmanSavaria1402, 2021-08-24:
> Still getting an error
> ImportError: No module named Extend.Topologyutils

It is TopologyUtils and not Topologyutils