tpaviot/pythonocc-core #559: installation issue under w64

Crashtest72, 2018-10-01:
Good morning,

I am trying to install python occ core 0.18.1 under W7 64:
-->anadonda navigator 1.9.2
-->python 3.6
-->Qt creator 4.7.0 based on qt5.11.1

I tried to install python occ with the conda link below:
conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1

I launch it by the terminal launched under the anaconda navigator

I get the following error each time I try:

""Executing transaction: failed
ERROR conda.core.link:_execute(502): An error occurred while uninstalling packag
e 'defaults::anaconda-5.2.0-py36_3'.
PermissionError(13, 'Accès refusé')""

Is there anybody to help me please ??
I have tried several way even with the source code to build, nothing works :((((

Thanks ahead.

Complete window below:
<<
(base) D:\Users\Nicolas.letard>conda install -c conda-forge -c dlr-sc -c pythono
cc -c oce pythonocc-core==0.18.1
Solving environment: done

## Package Plan ##

  environment location: C:\ProgramData\Anaconda3

  added / updated specs:
    - pythonocc-core==0.18.1

The following NEW packages will be INSTALLED:

    backports.functools_lru_cache: 1.5-py_1              conda-forge
    freeimageplus:                 3.17.0-vc14_0         dlr-sc      [vc14]
    gl2ps:                         1.3.8-vc14_3          dlr-sc      [vc14]
    krb5:                          1.14.6-vc14_0         conda-forge [vc14]
    oce:                           0.18.2-vc14h5f3c129_1 oce         [vc14]
    pythonocc-core:                0.18.1-py36_vc14_110  pythonocc   [vc14]
    smesh:                         6.7.5-vc14hfb33b19_0  pythonocc   [vc14]
    tbb:                           2017_20170226-vc14_0  conda-forge [vc14]

The following packages will be REMOVED:

    anaconda:                      5.2.0-py36_3

The following packages will be UPDATED:

    ca-certificates:               2018.03.07-0                      --> 2018.8.
24-ha4d7672_0 conda-forge
    certifi:                       2018.4.16-py36_0                  --> 2018.8.
24-py36_1001  conda-forge
    conda:                         4.5.11-py36_0                     --> 4.5.11-
py36_0        conda-forge
    curl:                          7.60.0-h7602738_0                 --> 7.60.0-
vc14_0        conda-forge [vc14]
    jpeg:                          9b-hb83a4c4_2                     --> 9b-vc14
_2            conda-forge [vc14]
    libsodium:                     1.0.16-h9d3ae62_0                 --> 1.0.16-
vc14_0        conda-forge [vc14]
    libxslt:                       1.1.32-hf6f1972_0                 --> 1.1.32-
vc14_0        conda-forge [vc14]
    lxml:                          4.2.1-py36heafd4d3_0              --> 4.2.3-p
y36heafd4d3_0 conda-forge
    openssl:                       1.0.2o-h8ea7d77_0                 --> 1.0.2p-
hfa6e2cd_1000 conda-forge
    tk:                            8.6.7-hcb92d03_3                  --> 8.6.8-v
c14_0         conda-forge [vc14]
    zeromq:                        4.2.5-hc6251cf_0                  --> 4.2.5-v
c14_2         conda-forge [vc14]

The following packages will be DOWNGRADED:

    bzip2:                         1.0.6-hfa6e2cd_5                  --> 1.0.6-v
c14_1         conda-forge [vc14]
    freetype:                      2.8-h51f8f2c_1                    --> 2.7-vc1
4_2           conda-forge [vc14]
    hdf5:                          1.10.2-hac2f561_1                 --> 1.10.2-
vc14_0        conda-forge [vc14]
    icu:                           58.2-ha66f8fd_1                   --> 58.2-vc
14_0          conda-forge [vc14]
    libiconv:                      1.15-h1df5818_7                   --> 1.14-vc
14_4          conda-forge [vc14]
    libpng:                        1.6.34-h79bbb47_0                 --> 1.6.28-
vc14_2        conda-forge [vc14]
    libtiff:                       4.0.9-hb8ad9f9_1                  --> 4.0.9-v
c14_0         conda-forge [vc14]
    libxml2:                       2.9.8-hadb2253_1                  --> 2.9.5-v
c14_1         conda-forge [vc14]
    lzo:                           2.10-h6df0209_2                   --> 2.10-vc
14_0          conda-forge [vc14]
    matplotlib:                    2.2.2-py36h153e9ff_1              --> 2.1.1-p
y36_0         conda-forge
    pillow:                        5.1.0-py36h0738816_0              --> 4.2.1-p
y36_0         conda-forge
    pyqt:                          5.9.2-py36h1aa27d4_0              --> 5.6.0-p
y36_2
    qt:                            5.9.5-vc14he4a7d60_0              --> 5.6.2-v
c14_1         conda-forge [vc14]
    snappy:                        1.1.7-h777316e_3                  --> 1.1.7-v
c14_1         conda-forge [vc14]
    sqlite:                        3.23.1-h35aae40_0                 --> 3.22.0-
vc14_0        conda-forge [vc14]
    yaml:                          0.1.7-hc54c509_2                  --> 0.1.7-v
c14_0         conda-forge [vc14]
    zlib:                          1.2.11-h8395fce_2                 --> 1.2.11-
vc14_0        conda-forge [vc14]

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: failed
ERROR conda.core.link:_execute(502): An error occurred while uninstalling packag
e 'defaults::anaconda-5.2.0-py36_3'.
PermissionError(13, 'Accès refusé')
Attempting to roll back.

Rolling back transaction: done

PermissionError(13, 'Accès refusé')

(base) D:\Users\Nicolas.letard>
>>

Crashtest72, 2018-10-02:
Is just possible to know if at least the softwares I use are compatible with the installation of python occ?

efirvida, 2018-10-29:
did you solve this?

Crashtest72, 2018-10-29:
Not yet ... :(

johannesmichael, 2018-11-21:
Hi
I have a similiar issue, I opened in pythonocc-demos. 
I will close it there

After
`conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1`

conda env cannot be used any more.

Even for conda list, I get

`system cannot find the path specified`

Any suggestions what causes this?

Regards
johannes

System: Win 10
conda version : 4.5.11
conda-build version : 3.4.1
python version : 3.6.6

johanesmikhael, 2018-11-23:
> Hi
> I have a similiar issue, I opened in pythonocc-demos.
> I will close it there
>
> After
> `conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1`
>
> conda env cannot be used [...]

Hi, i came accross withe the same issue few weeks ago but in different circumstance. Check my answer [here](
https://github.com/conda/conda/issues/344#issuecomment-435561895). It has something to do with the files in conda Script folder.