tpaviot/pythonocc-core #563: Python3 installation

rsm-gh, 2018-10-19:
Hello,

I have some questions concerning the python3 installation. Some years ago I was able of doing but with the new releases I'm really struggling on Debian...

I think that the problem is that I do not have the flag for the installation directory but there may be other problems.

I'm compiling and installing `OCE` with the following command:

`cmake -DOCE_INSTALL_PREFIX:PATH=/usr/bin/oce ..`

and it seems okay (the tests and installation is alright):

```
root@castor-dev:/home/rsm/Desktop/pythonocc-core-0.18.1/build# whereis oce
oce: /usr/bin/oce
root@castor-dev:/home/rsm/Desktop/pythonocc-core-0.18.1/build# ls -la /usr/bin/oce
total 64
drwxr-xr-x 5 root root  4096 Oct 17 03:36 .
drwxr-xr-x 3 root root 36864 Oct 19 09:49 ..
drwxr-xr-x 3 root root  4096 Oct 17 03:36 include
drwxr-xr-x 3 root root 12288 Oct 17 03:36 lib
drwxr-xr-x 3 root root  4096 Oct 17 03:36 share
```

Then I try compiling OCC with the following flags:
```

cmake 
-DPYTHON_EXECUTABLE:PATH=/usr/lib/python3.5 
-DPYTHON_LIBRARY:PATH=/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0
-PYTHONOCC_INSTALL_DIRECTORY:PATH=/usr/lib/python3/dist-packages/OCC
-DOCE_INCLUDE_PATH=/usr/bin/oce/include/oce
-DOCE_LIB_PATH=/usr/bin/oce/lib  ..
```

but the `-PYTHONOCC_INSTALL_DIRECTORY:`  flag makes crash `cmake`. If I remove the flag, `cmake` seems to work alright:

```
root@castor-dev:/home/rsm/Desktop/pythonocc-core-0.18.1/build# cmake -DPYTHON_EXECUTABLE:PATH=/usr/lib/python3.5 -DPYTHON_LIBRARY:PATH=/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0 -DOCE_INCLUDE_PATH=/usr/bin/oce/include/oce -DOCE_LIB_PATH=/usr/bin/oce/lib  ..
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
--  
-- Found PythonInterp: /usr/lib/python3.5 (found version "1.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0 (found version "3.5.3") 
-- Build 64bit
-- Found SWIG: /usr/bin/swig3.0 (found suitable version "3.0.10", minimum required is "3.0.9") 
-- Found Freetype: /usr/lib/x86_64-linux-gnu/libfreetype.so (found version "2.6.3") 
-- Platform:Unix
-- output_path: Unix/x86_64-MinSizeRel-64
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rsm/Desktop/pythonocc-core-0.18.1/build
```

Then the compilation is alright, but the installation directory is `/`. Anyways in order to test OCC I created the following python script:

```
import sys
sys.path.insert(0, "/")

from OCC.STEPControl import STEPControl_Reader
```
and added an `__init__.py` file in root. But I still getting the following error: 
```

rsm@castor-dev:~/Desktop$ python3 test1.py
Traceback (most recent call last):
  File "/OCC/STEPControl.py", line 24, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 914, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: libTKernel.so.11: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test1.py", line 6, in <module>
    from OCC.STEPControl import STEPControl_Reader
  File "/OCC/STEPControl.py", line 27, in <module>
    _STEPControl = swig_import_helper()
  File "/OCC/STEPControl.py", line 26, in swig_import_helper
    return importlib.import_module('_STEPControl')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: No module named '_STEPControl'
```

Which as I was saying, I believe that it is cause because of the installation path. When making the installation of OCC I got:

```
...]
Install the project...
-- Install configuration: "MinSizeRel"
-- Installing: /OCC/FSD.py
-- Installing: /OCC/_FSD.so
-- Set runtime path of "/OCC/_FSD.so" to ""
-- Installing: /OCC/MMgt.py
-- Installing: /OCC/_MMgt.so
-- Set runtime path of "/OCC/_MMgt.so" to ""
-- Installing: /OCC/OSD.py
-- Installing: /OCC/_OSD.so
-- Set runtime path of "/OCC/_OSD.so" to ""
-- Installing: /OCC/Plugin.py
-- Installing: /OCC/_Plugin.so
-- Set runtime path of "/OCC/_Plugin.so" to ""
-- Installing: /OCC/Quantity.py
-- Installing: /OCC/_Quantity.so
[...
```

Can some one give me the right flag to specify the OCC directory? And In case that the error be caused because another reason any help would be really appreciated !!

Thanks !

rsm-gh, 2018-10-23:
by using the cmake command `cmake -LA | awk '{if(f)print} /-- Cache values/{f=1}'` I got the missing flags and I corrected them:

```
cmake
-DPYTHON_EXECUTABLE:PATH=/usr/lib/python3.5
-DPYTHON_INCLUDE_DIR:PATH=/usr/include/python3.5
-DPYTHON_LIBRARY:PATH=/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0
-DPYTHONOCC_INSTALL_DIRECTORY:PATH=/usr/lib/python3.5/dist-packages/OCC
-DOCE_INCLUDE_PATH=/usr/bin/oce/include/oce -DOCE_LIB_PATH=/usr/bin/oce/lib  ..
```

Everything seems alright...:

```
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
--  
-- Found PythonInterp: /usr/lib/python3.5 (found version "1.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0 (found version "3.5.3") 
-- Build 64bit
-- Found SWIG: /usr/bin/swig3.0 (found suitable version "3.0.10", minimum required is "3.0.9") 
-- Found Freetype: /usr/lib/x86_64-linux-gnu/libfreetype.so (found version "2.6.3") 
-- Platform:Unix
-- output_path: Unix/x86_64-MinSizeRel-64
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Configuring done
-- Generating done
```

```
....]
FREETYPE_INCLUDE_DIR_freetype2:PATH=/usr/include/freetype2
FREETYPE_INCLUDE_DIR_ft2build:PATH=/usr/include/freetype2
FREETYPE_LIBRARY_DEBUG:FILEPATH=FREETYPE_LIBRARY_DEBUG-NOTFOUND
FREETYPE_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libfreetype.so
OCE_INCLUDE_PATH:PATH=/usr/bin/oce/include/oce
OCE_LIB_PATH:PATH=/usr/bin/oce/lib
OPENGL_INCLUDE_DIR:PATH=/usr/include
OPENGL_gl_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libGL.so
OPENGL_glu_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libGLU.so
OPENGL_xmesa_INCLUDE_DIR:PATH=OPENGL_xmesa_INCLUDE_DIR-NOTFOUND
PYTHONOCC_BUILD_TYPE:STRING=MinSizeRel
PYTHONOCC_INSTALL_DIRECTORY:PATH=/usr/lib/python3.5/dist-packages/OCC
PYTHONOCC_WRAP_DATAEXCHANGE:BOOL=ON
PYTHONOCC_WRAP_OCAF:BOOL=ON
PYTHONOCC_WRAP_SMESH:BOOL=OFF
PYTHONOCC_WRAP_VISU:BOOL=ON
PYTHON_EXECUTABLE:PATH=/usr/lib/python3.5
PYTHON_INCLUDE_DIR:PATH=/usr/include/python3.5
PYTHON_LIBRARY:PATH=/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1.0
PYTHON_LIBRARY_DEBUG:FILEPATH=PYTHON_LIBRARY_DEBUG-NOTFOUND
SWIG_DIR:PATH=/usr/share/swig3.0
SWIG_EXECUTABLE:FILEPATH=/usr/bin/swig3.0
SWIG_VERSION:STRING=3.0.10

```

but OCC stills having troubles to find OCE:

```
rsm@castor-dev:~$ python3.5
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from OCC.STEPControl import STEPControl_Reader
Traceback (most recent call last):
  File "/usr/lib/python3.5/dist-packages/OCC/STEPControl.py", line 24, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 577, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 914, in create_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
ImportError: libTKernel.so.11: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.5/dist-packages/OCC/STEPControl.py", line 27, in <module>
    _STEPControl = swig_import_helper()
  File "/usr/lib/python3.5/dist-packages/OCC/STEPControl.py", line 26, in swig_import_helper
    return importlib.import_module('_STEPControl')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: No module named '_STEPControl'
```

**libTKernel.so.11** is not found but it is actually at the right place:
```

rsm@castor-dev:~$ ls -la /usr/bin/oce/lib | grep TKer
lrwxrwxrwx 1 root root      17 Oct 17 03:36 libPTKernel.so -> libPTKernel.so.11
lrwxrwxrwx 1 root root      21 Oct 17 03:36 libPTKernel.so.11 -> libPTKernel.so.11.0.0
-rw-r--r-- 1 root root  391120 Oct 17 03:18 libPTKernel.so.11.0.0
lrwxrwxrwx 1 root root      16 Oct 17 03:36 libTKernel.so -> libTKernel.so.11
lrwxrwxrwx 1 root root      20 Oct 17 03:36 libTKernel.so.11 -> libTKernel.so.11.0.0
-rw-r--r-- 1 root root 2650000 Oct 17 03:18 libTKernel.so.11.0.0
```

Any idea of why the `OCE_LIB_PATH:PATH=/usr/bin/oce/lib` is not working correctly?

jf---, 2018-10-23:
Why not try `conda` as suggested on the README.md?

rsm-gh, 2018-10-23:
@jf---  well because I suppose that conda is part of Anaconda, meaning that I'll have to install Anaconda on Debian.

I don't think that it be a proper solution since Anaconda includes too much stuff and also because I will have to replace the default python of the server. To be honest I'd rather passing the server in to ArchLinux if that's necessary (I have no problem doing the install on Arch).

I also think that having a python 3.X compilation guide would be useful for users like me that want or need to compile the sources.

rsm-gh, 2018-10-26:
I just made it work by using the default location for python OCE and modifying the OCC include path. 

It seems strange that when using the other OCE locations it didn't work since OCC had the good folder, but since it works I'm going to close the issue.

Good luck for another person having the same problem !!

closed 2018-10-26 by rsm-gh