tpaviot/pythonocc-core #568: SetTextureRepeat() Boolean flag appears to be broken

jrj99, 2018-10-31:
For a recent project, I was trying to disable texture repeat in order to get a texture to occupy part of a face without repeating. however, I could not get this to work, and I believe that the boolean flag for setTextuerRepeat may be broken. This is demonstrated by the code below. Notice that even though texture repeatU and texture repeatV are set to 3.0, the boolean flag in setTextureRepeat is false, so these values should be nullified. However, they are not. The texture still repeats three times per face. 
My code is below, and the image output is below as well.

```
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.AIS import AIS_Shape
import OCC.Quantity
from OCC.Display.SimpleGui import init_display
from OCC.V3d import (V3d_ZBUFFER, V3d_PHONG, V3d_Zpos, V3d_Zneg, V3d_Xpos,
                     V3d_Xneg, V3d_Ypos, V3d_Yneg, V3d_XposYnegZpos, V3d_TEX_ALL,
                     V3d_TEX_NONE, V3d_TEX_ENVIRONMENT,
                     V3d_LayerMgr)
from OCC.AIS import AIS_Shape, AIS_Shaded, AIS_TexturedShape, AIS_WireFrame
from OCC.TCollection import TCollection_ExtendedString, TCollection_AsciiString
from OCC.Extend.TopologyUtils import TopologyExplorer
import DataExchange

#enable texture mapping
def texture_faces(a_face):
    transp = 0.6
    toScaleU = 1.0
    toScaleV = 1.0
    toRepeatU = 3.0
    toRepeatV = 3.0
    originU = 1.0
    originV = 1.0

    display.View.SetSurfaceDetail(V3d_TEX_ALL)
    aisShape = AIS_TexturedShape(a_face)
    aisShape.SetTextureFileName(TCollection_AsciiString('C:/Projects/pyOCC_work/wooden_crate_texture.png'))
    aisShape.SetTextureMapOn()
    aisShape.SetTextureScale(False, toScaleU, toScaleV)
    aisShape.SetTextureRepeat(False, toRepeatU, toRepeatV)
    aisShape.SetTextureOrigin(False, originU, originV)
    #aisShape.EnableTextureModulate()
    aisShape.SetDisplayMode(3)
    h_aisShape = aisShape.GetHandle()
    display.Context.Display(aisShape.GetHandle(), False)

if __name__ == '__main__':
[](url)
    display, start_display, add_menu, add_function_to_menu = init_display()
    myBox = BRepPrimAPI_MakeBox(60,60,50).Shape()
    texture_faces(myBox)
    #myBox = DataExchange.read_stl_file('knob1.stl')
    display.View_Iso()
    display.FitAll()
    start_display()
```
[image]

tpaviot, 2018-10-31:
Can u please attache the png file in order to reproduce the issue

jrj99, 2018-10-31:
Sure. Here is the PNG file I used.
[image]

jrj99, 2018-10-31:
printing aisShape.TextureRepeat() in this snippet of code displays "false" in the console. 

```
    aisShape.SetTextureScale(False, toScaleU, toScaleV)
    aisShape.SetTextureRepeat(False, toRepeatU, toRepeatV)
    aisShape.SetTextureOrigin(False, originU, originV)
    aisShape.SetDisplayMode(3)
    print(aisShape.TextureRepeat())
```

jrj99, 2018-11-02:
> If I set toRepeatU and to RepeatV to 1.0, I get the following:
>
> [image: texture_scr]
> <https://user-images.githubusercontent.com/660130/47927705-540e9c00-dec4-11e8-87ac-ec36e3ca33f2.png>
>
> Isn't that what you want ?

I was hoping to map a small version of the texture on a face (using the
scale parameters) and map the texture so that the texture only took up
1\4th of a face. I want the other 3/4 of the face to be shaded without any
textures present. According to Open Cascade's documentation, I can turn off
texture repeating with a boolean command, which should allow me to
accomplish what I want. However I couldn't set texture repeat to false in
python OCC.

-Jared

tpaviot, 2024-05-23:
outdated issue, please reopen if still valid

closed 2024-05-23 by tpaviot