tpaviot/pythonocc-core #576: OpenCASCADE 7.2.0 support

rainman110, 2018-11-12:
I have seen, there is a branch to support OpenCASCADE 7.2.0. What is the current status?

Do you need any help?

tpaviot, 2018-11-13:
@rainman110 I started working on occt720/pyocc support a few weeks before the summer. Then I got absorbed in another project I'm just leaving out. I'm reviewing all issue tickets I missed, from newer to older, still have to review the Handle-* ticket you opened. After that I can get back to occt7 support, I guess 7.3 is a better choice now

rainman110, 2018-11-13:
I once tried (for one day) to bind opencascade 7.2.0 but failed mainly due to the new Container classes that are now based on std containers.

So if you already have a good starting point, I could coninue on that. So, what is already working and what remains to be done?

The Handle* stuff could be integrated into the major pythonocc release. I have a working copy in my own github account and played with that. It works pretty well. Removing the Handle is definitely better for the users and also for the IDEs, as it will also improve auto-completion.

tpaviot, 2018-11-13:
I don't remember exactly what's working and what's not, I spent 2 weeks on that I have to check the current status of this branch.

For the Handle_* stuff, if that works please create a pull request. There's already an API change (the move from OCC to OCC.Core), so it's a good time to insert another change.

rainman110, 2018-11-13:
One more question: Your changes to make OCCT 7.2.0 work don't seem to be made with the generator or do they? How do you process, if you have to make changes by hand to the .i files?

tpaviot, 2018-11-14:
Here is the workflow for migrating to higher occt version:

1. generate all the .i SWIG files from the OCCT headers, using the pythonocc-generator

2. try to compile only a few basic modules (mostly Standard, gp, BRepPrimAPI) ;

3 Of course i does not compile. Then I manually fix SWIG files so that they compile.

4. When I fully understood what changes are need, I change the generator accordingly so that those changes are made available for all modules

5. Regenerate the modules.

6. Back to 2

rainman110, 2019-11-14:
Closing this in favor of OpenCASCADE 7.4.0

closed 2019-11-14 by rainman110