tpaviot/pythonocc-core #580: Getting STEP reader progress

thomas-lamure, 2018-11-14:
In OCC there is a method called `SetProgress` under `MapReader` in `WS` of `Reader`.

Trying to access this object i got this Warning :
`swig/python detected a memory leak of type 'Handle_Transfer_TransientProcess *', no destructor found.`

And after, trying to call SetProgress, I got this Error :
`AttributeError: 'SwigPyObject' object has no attribute 'SetProgress'`

My code is :
```python
step_reader = STEPCAFControl_Reader()
step_reader.Reader().WS().GetObject().MapReader().SetProgress(progress_indicator)
```

Have you ever used a `ProgressIndicator` and how ? and is the Warning linked to the problem ?

Thank-you.

jf---, 2018-11-15:
what object is `.MapReader()`?

tpaviot, 2018-11-15:
A few hints about message indicators:

https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Import/Gui/AppImportGuiPy.cpp#L413

http://opencascade.blogspot.com/2009/03/unnoticeable-memory-leaks-part-2.html

I never worked with Message_ProgressIndicator class. It seems it has to be subclassed, but I don't know if the specialization can be done at the python level.

@thomas-lamure The Transfer module is not wrapped in the latest pythonocc-core released. It was added a few days ago see commit ca93e887bc2f1587798d65181140b679aa57c900

Rebuild your local copy or install/download one of the nightly build available at https://anaconda.org/tpaviot/pythonocc-core/

thomas-lamure, 2018-11-15:
OK, thank you for your help.

I'll look into it.