tpaviot/pythonocc-core #586: 关于使用pythonocc 18.2  把igs文件转换成 stl文件时出现转换出来的文件特别大的一个问题。

xiaojianbao, 2018-11-24:
Hello, when I use pythons to convert to stl file when using pythonocc 18.2, the conversion file becomes very big problem. My original file is 2.8M, but the size of the converted stl file reaches 1.18G. My test program and igs file.

The following is my test program

[image]

import sys
from OCC.Extend.DataExchange import (read_step_file, write_step_file, read_stl_file, write_stl_file ,read_iges_file, write_iges_file)
from OCC.Extend.ShapeFactory import (get_boundingbox)
# step to stl
shape_to_return = read_iges_file("D:////xiaojianbao//pythonocc-file//jz.igs")
print("Data reading completed")
result_status = write_stl_file(shape_to_return, "D:////xiaojianbao//pythonocc-file//jz0.6.stl", "ascii", 0.9, 0.5)
print("Data write completed")

My test file
[test.zip](https://github.com/tpaviot/pythonocc-core/files/2612126/test.zip)

rainman110, 2018-11-28:
You might need to increase the deflection parameter to get fewer triangles.

xiaojianbao, 2018-11-29:
> You might need to increase the deflection parameter to get fewer triangles.
Hello, I feel your reply very much. I set the triangle parameter to 4.5 and 2.5 but the converted file still has more than 700 M.

liuxin2322, 2018-11-30:
老哥，我看到你使用了igs文件，可以询问一下问题么？（我个人还在学校读书，现在在做一个项目需要这个）
就igs读取后，鼠标可以选中面，获取umin umax vmin vmax 
这用什么函数去做呀？以及这个函数怎么用的？
如果能回答我非常感谢了！

xiaojianbao, 2018-11-30:
> 老哥，我看到你使用了igs文件，可以询问一下问题么？（我个人还在学校读书，现在在做一个项目需要这个）
> 就igs读取后，鼠标可以选中面，获取umin umax vmin vmax
> 这用什么函数去做呀？以及这个函数怎么用的？
> 如果能回答我非常感谢了！
我也不清楚怎么用这些函数，以前都没有接触过python

rainman110, 2018-11-30:
How large is your model in terms of geometric units?

xiaojianbao, 2018-11-30:
[image]

xiaojianbao, 2018-11-30:
volume = 67604.146
area = 41626.8.8
length * width *  height = 124.50 x 50.00 x 36.30 mm

tpaviot, 2018-12-04:
@xiaojianbao export to binary, it will be lighter. If you can upload your igs file, it will be easier to check what's going on

xiaojianbao, 2018-12-05:
> @xiaojianbao export to binary, it will be lighter. If you can upload your igs file, it will be easier to check what's going on

In the bottom of the post, I have uploaded the igs file I tested. In fact, I found that not only the conversion is very slow, but also when reading the data, it is very slow, and not only this one file, it is very big. Some igs files have this problem.

xiaojianbao, 2018-12-05:
@tpaviot In the bottom of the post, I have uploaded the igs file I tested. In fact, I found that not only the conversion is very slow, but also when reading the data, it is very slow, and not only this one file, it is very big. Some igs files have this problem.

jzuefle, 2019-03-08:
Hello, getting the same issue but exporting back to iges.
Is there a reason in DataExchange read_iges_file the nbr loop is necessary?
Since it squares the number of faces of all my loaded cad files. Can't see anything doing with the compound.