tpaviot/pythonocc-core #591: get_qt_modules() return error from qtDisplay.py

zhaowei699, 2018-12-07:
Traceback (most recent call last):
  File "qtDisplay3.py", line 29, in <module>
    QtCore, QtGui, QtWidgets, QtOpenGL = get_qt_modules()
  File "C:\software\anaconda3440\envs\pythonocc\lib\site-packages\OCC\Display\backend.py", line 238, in get_qt_modules
    raise ValueError("no backend has been imported yet with "
ValueError: no backend has been imported yet with ``load_backend``...

I installed pythonocc-core by this method:
(base) C:\Users\wei.zhao>conda create -n pythonocc -c dlr-sc pythonocc-core
Solving environment: done

## Package Plan ##

  environment location: C:\software\anaconda3440\envs\pythonocc

  added / updated specs:
    - pythonocc-core

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    icu-58.2                   |       ha66f8fd_1        21.9 MB
    wheel-0.31.1               |           py35_0          81 KB
    zlib-1.2.11                |       h62dcd97_3         128 KB
    libpng-1.6.35              |       h2a8f88b_0         589 KB
    sqlite-3.25.3              |       he774522_0         937 KB
    pyqt-5.9.2                 |   py35h6538335_2         4.2 MB
    sip-4.19.8                 |   py35h6538335_0         282 KB
    pythonocc-core-0.17        |      py35_vc14_4        39.6 MB  dlr-sc
    vs2015_runtime-14.15.26706 |       h3a45250_0         2.2 MB
    wincertstore-0.2           |   py35hfebbdb8_0          13 KB
    python-3.5.6               |       he025d50_0        18.2 MB
    setuptools-40.2.0          |           py35_0         597 KB
    pip-10.0.1                 |           py35_0         1.8 MB
    certifi-2018.8.24          |           py35_1         140 KB
    ca-certificates-2018.03.07 |                0         155 KB
    qt-5.9.6                   |   vc14h1e9a669_2        93.9 MB
    openssl-1.0.2p             |       hfa6e2cd_0         5.4 MB
    vc-14.1                    |       h0510ff6_4           6 KB
    ------------------------------------------------------------
                                           Total:       190.1 MB

The following NEW packages will be INSTALLED:

    ca-certificates: 2018.03.07-0
    certifi:         2018.8.24-py35_1
    freeimageplus:   3.17.0-vc14_0          dlr-sc [vc14]
    freetype:        2.6.3-vc14_1           dlr-sc [vc14]
    gl2ps:           1.3.8-vc14_3           dlr-sc [vc14]
    icu:             58.2-ha66f8fd_1
    jpeg:            9b-vc14_0              dlr-sc [vc14]
    libpng:          1.6.35-h2a8f88b_0
    oce:             0.17.2-vc14_7          dlr-sc [vc14]
    openssl:         1.0.2p-hfa6e2cd_0
    pip:             10.0.1-py35_0
    pyqt:            5.9.2-py35h6538335_2
    python:          3.5.6-he025d50_0
    pythonocc-core:  0.17-py35_vc14_4       dlr-sc [vc14]
    qt:              5.9.6-vc14h1e9a669_2
    setuptools:      40.2.0-py35_0
    sip:             4.19.8-py35h6538335_0
    sqlite:          3.25.3-he774522_0
    tbb:             4.3.6-vc14_1           dlr-sc [vc14]
    vc:              14.1-h0510ff6_4
    vs2015_runtime:  14.15.26706-h3a45250_0
    wheel:           0.31.1-py35_0
    wincertstore:    0.2-py35hfebbdb8_0
    zlib:            1.2.11-h62dcd97_3

liuxin2322, 2018-12-07:
can you show error figure?有出bug的图么？

zhaowei699, 2018-12-10:
I updated the function "get_qt_modules()" to solve, because I only use pyqt5, so I don't care others.
`def get_qt_modules():`
`  `    `  # add by wei`
  `  `    ` from PyQt5 import QtCore, QtGui, QtOpenGL, QtWidgets`
 `    `    `return QtCore, QtGui, QtWidgets, QtOpenGL`
 `   `    ` # end adding by wei`
thank you very much

tpaviot, 2019-01-13:
Bug cannot be reproduced. pythonocc-core 0.17 is outdated. Please upgrade to 0.18.1 or, even better, to the 0.18.2 nightly builds at https://anaconda.org/tpaviot/pythonocc-core

closed 2019-01-13 by tpaviot