tpaviot/pythonocc-core #594: Issue with running examples using OCC.DataExchange after conda install

markbasham, 2018-12-07:
Hi,

Cool project, really looking forward to using it :) I have though had a couple of issues with running the examples, and any help would be really appreciated.

I installed the project using the conda install in python 3.6.6 running on Window 10
`$ conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1`

Currently after replacing OCC.core import with OCC I can get the hello world example working nicely, but then I tried the core_load_step_with_colors.py example and get the following error.

```
(base) C:\Users\ssg37927\Documents\GitHub\pythonocc-demos\examples>python core_load_step_with_colors.py
Traceback (most recent call last):
  File "core_load_step_with_colors.py", line 18, in <module>
    from OCC.DataExchange import read_step_file_with_names_colors
ModuleNotFoundError: No module named 'OCC.DataExchange'
```

Thanks, Mark

markbasham, 2018-12-07:
Sorry I just realised I had already had a quick go at fixing the code and changed the import, the real error is this
```
(base) C:\Users\ssg37927\Documents\GitHub\pythonocc-demos\examples>python core_load_step_with_colors.py
Traceback (most recent call last):
  File "core_load_step_with_colors.py", line 18, in <module>
    from OCC.Extend.DataExchange import read_step_file_with_names_colors
ModuleNotFoundError: No module named 'OCC.Extend'
```

markbasham, 2018-12-12:
@tpaviot Any idea what the problem here could be, I'm trying to write both step and stl files, step works fine, but stl relies on the extend package.  As a workaround can i just grab the repo and put it on my pythonpath, or is it likely i will need to build something?

beeekey, 2019-01-11:
Having the same issue after installing with the exact same conda command

tpaviot, 2019-01-11:
There is an api change in the current pythonocc master branch, this change as propagated to pythonocc-demos.  You can test a nightly build here:
```$ conda install -c conda-forge -c dlr-sc -c tpaviot -c oce pythonocc-core==0.18.2```

Next release is planned for late february

msato, 2019-01-11:
Hi.
I tried conda install. but I could not find package.

(base) C:\Users\sato>conda install -c conda-forge -c dlr-sc -c tpaviot -c oce pythonocc-core==0.18.2
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - pythonocc-core==0.18.2
What should I do?
Thanks.

msato, 2019-01-11:
I retried. but I got same result.

tpaviot, 2019-01-11:
@msato is your platform win32/py27 ?

msato, 2019-01-12:
my platform is win64/py3.6.

msato, 2019-01-12:
I downloaded pythonocc-core-0.18.2-py36_vc14h24bf2e0_220.tar.bz2 from https://anaconda.org,
then I installed it as below.

(C:\Users\sato\Anaconda3-440) C:\Users\sato>conda install pythonocc-core-0.18.2-py36_vc14h24bf2e0_220.tar.bz2

Downloading and Extracting Packages
############################################################################################################### | 100%
Preparing transaction: done
Verifying transaction: done
Executing transaction: done

It's done well.
Thanks.

closed 2019-01-12 by tpaviot

beeekey, 2019-01-12:
Thank you very much, I was not able to use your command, but with the following, it worked:

`conda install -c tpaviot/label/cd-master pythonocc-core`

I don't understand why, but it works now pretty fine. 

Thanks.

tpaviot, 2019-01-13:
It's because I updated/uploaded all conda packages to anaconda