tpaviot/pythonocc-core #595: bug: about qtViewer3d

zhaowei699, 2018-12-10:
Hi
If you use pyqt5, please kindly run the scipt SimpleGui.py(../Lib/site-packages/OCC/Display/SimpleGui.py) directly. 
It looks good to open a window, but if you try to press "Shift" and press left-button of mouse to draw a rectangle(in order to select some objects), the script will be crashed. 
(no error information, the window will be destroyed directly)

I create some points and I want to select them by "Shift" + left-button of mouse. If this bug can't be solved, I can't continue to do next step. Please kindly help me.

zhaowei699, 2018-12-13:
I found if changed "class qtBaseViewer(QtOpenGL.QGLWidget):" to "class qtBaseViewer(QtWidgets.QOpenGLWidget):" in qtDisplay.py, it can solve this bug. If you use high version (above pyqt5.6)

waterbug, 2018-12-13:
This also works in pyqt 5.6:

     class QtBaseViewer(QtOpenGL.QGLWidget)

waterbug, 2018-12-13:
Sorry, never mind, I just realized you were saying "higher than 5.6"!
(Need more coffee ... :P)

tpaviot, 2019-01-14:
@zhaowei699 please report pythonocc-core version you've been using. 0.17 ? 0.18.1 ? 

Fixed in current master, successfully tested with pyqt-5. 6 and 5.9, check nightly builds

closed 2019-01-14 by tpaviot