tpaviot/pythonocc-core #600: Segafault on Windows and 18.2

adam-urbanczyk, 2018-12-11:
I am trying out 18.2 for Cadquery https://github.com/CadQuery/cadquery/pull/57 but getting a segfault while using `HLRBRep_HLRToShape` and some methods of the resulting object.  Any clues how to solve it?

FYI: everything works well on linux.

tpaviot, 2018-12-12:
Hard to say without a test case

adam-urbanczyk, 2018-12-13:
Yes, here it is:

```python
from OCC.Core.gp import gp_Ax2, gp_Pnt, gp_Dir
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.HLRBRep import HLRBRep_Algo, HLRBRep_HLRToShape
from OCC.Core.HLRAlgo import HLRAlgo_Projector

box = BRepPrimAPI_MakeBox(1.,1.,1.).Shape()

hlr = HLRBRep_Algo()
hlr.Add(box)

projector = HLRAlgo_Projector(gp_Ax2(gp_Pnt(),gp_Dir(-1.75, 1.1, 5)))

hlr.Projector(projector)
hlr.Update()
hlr.Hide()

hlr_shapes = HLRBRep_HLRToShape(hlr.GetHandle())

visible_smooth_edges = hlr_shapes.Rg1LineVCompound()

print(visible_smooth_edges)

```
This is what I see in WinDbg
```
(256c.2b00): Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
*** WARNING: Unable to verify checksum for C:\Localdata\Anaconda3_64\envs\cqocc\lib\site-packages\OCC\Core\_HLRBRep.pyd
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Localdata\Anaconda3_64\envs\cqocc\lib\site-packages\OCC\Core\_HLRBRep.pyd - 
_HLRBRep!gp_Ax3::XReverse+0x8d3c7:
00007ff9`30f805e1 488b01          mov     rax,qword ptr [rcx] ds:00000000`00000000=????????????????
```

And for dessert my conda env:
```
asn1crypto                0.24.0                py36_1003    conda-forge
backcall                  0.1.0                     <pip>
ca-certificates           2018.11.29           ha4d7672_0    conda-forge
certifi                   2018.11.29            py36_1000    conda-forge
cffi                      1.11.5          py36hfa6e2cd_1001    conda-forge
chardet                   3.0.4                 py36_1003    conda-forge
codecov                   2.0.15                     py_1    conda-forge
colorama                  0.4.1                     <pip>
coverage                  4.5.2           py36hfa6e2cd_1000    conda-forge
cryptography              2.3.1           py36h74b6da3_1000    conda-forge
decorator                 4.3.0                     <pip>
freeimageplus             3.17.0                   vc14_0  [vc14]  dlr-sc
freetype                  2.9.1             he8b6a0d_1004    conda-forge
gl2ps                     1.3.8                    vc14_3  [vc14]  dlr-sc
idna                      2.8                   py36_1000    conda-forge
ipdb                      0.11                      <pip>
ipython                   7.2.0                     <pip>
ipython-genutils          0.2.0                     <pip>
jedi                      0.13.1                    <pip>
libpng                    1.6.36            h7602738_1000    conda-forge
mock                      2.0.0                 py36_1000    conda-forge
oce                       0.18.3                   vc14_3  [vc14]  oce
openssl                   1.0.2p            hfa6e2cd_1001    conda-forge
parso                     0.3.1                     <pip>
pbr                       4.2.0                      py_0    conda-forge
pickleshare               0.7.5                     <pip>
pip                       18.1                  py36_1000    conda-forge
prompt-toolkit            2.0.7                     <pip>
pycparser                 2.19                       py_0    conda-forge
Pygments                  2.3.0                     <pip>
pyopenssl                 18.0.0                py36_1000    conda-forge
pyparsing                 2.3.0                      py_0    conda-forge
pysocks                   1.6.8                 py36_1002    conda-forge
python                    3.6.6                he025d50_0    conda-forge
pythonocc-core            0.18.2          py36_vc14h24bf2e0_183  [vc14]  tpaviot
requests                  2.21.0                py36_1000    conda-forge
setuptools                40.6.2                   py36_0    conda-forge
six                       1.12.0                py36_1000    conda-forge
smesh                     6.7.6                    vc14_7  [vc14]  pythonocc
tbb                       2019.3            he980bc4_1000    conda-forge
traitlets                 4.3.2                     <pip>
urllib3                   1.24.1                py36_1000    conda-forge
vc                        14                            0    conda-forge
vs2015_runtime            14.0.25420                    0    conda-forge
wcwidth                   0.1.7                     <pip>
wheel                     0.32.3                   py36_0    conda-forge
win_inet_pton             1.0.1                 py36_1002    conda-forge
wincertstore              0.2                   py36_1002    conda-forge
zlib                      1.2.11            h2fa13f4_1003    conda-forge
```

tpaviot, 2018-12-13:
No problem on my Windows machine, I can't reproduce the issue.

There was a recent big API change : Handles have disapeear. Try to replace ``hlr_shapes = HLRBRep_HLRToShape(hlr.GetHandle())`` with ``hlr_shapes = HLRBRep_HLRToShape(hlr)``

adam-urbanczyk, 2018-12-14:
I tried it and still no change. Are you sure that you get to the print statement? BTW: I also get a return code of `-1073741819`. I also observe same issue in appveyor.

tpaviot, 2018-12-14:
No, I don't reach the print statement, program stops before but I don't get any sefault. It fails at ```hlr_shapes = HLRBRep_HLRToShape(hlr.GetHandle())```. Did it use to work with current release (0.18.1) ?

tpaviot, 2018-12-14:
It does not work either with 0.18.1

tpaviot, 2018-12-14:
@adam-urbanczyk did you try with other shapes different from a single box ?

adam-urbanczyk, 2018-12-20:
Yes, same issue with e.g. the following shape  [Uploading out.zip…]()

I am going to build a debug binary of 18.2 and provide more details

adam-urbanczyk, 2018-12-20:
OK, this is what I have so far. It goes wrong at the very end of the pasted code.

```c++
SWIGINTERN PyObject *_wrap_HLRBRep_HLRToShape_Rg1LineVCompound__SWIG_0(PyObject *SWIGUNUSEDPARM(self), int nobjs, PyObject **swig_obj) {
  PyObject *resultobj = 0;
  HLRBRep_HLRToShape *arg1 = (HLRBRep_HLRToShape *) 0 ;
  void *argp1 = 0 ;
  int res1 = 0 ;
  TopoDS_Shape result;
  
  if ((nobjs < 1) || (nobjs > 1)) SWIG_fail;
  res1 = SWIG_ConvertPtr(swig_obj[0], &argp1,SWIGTYPE_p_HLRBRep_HLRToShape, 0 |  0 );
  if (!SWIG_IsOK(res1)) {
    SWIG_exception_fail(SWIG_ArgError(res1), "in method '" "HLRBRep_HLRToShape_Rg1LineVCompound" "', argument " "1"" of type '" "HLRBRep_HLRToShape *""'"); 
  }
  arg1 = reinterpret_cast< HLRBRep_HLRToShape * >(argp1);
  {
    try
    {
      OCC_CATCH_SIGNALS
      result = (arg1)->Rg1LineVCompound();
    } 
    catch(Standard_Failure)
    {
      Handle(Standard_Failure) error = Standard_Failure::Caught ();
      char *error_name = (char*) error->DynamicType()->Name();
      char *error_message = (char*) error->GetMessageString();
      std::string message;
      if (error_name) message += std::string(error_name) + "\n";
      if (error_message) message += std::string(error_message);
      // raise the python exception
      PyErr_SetString(PyExc_RuntimeError, message.c_str());
      return NULL;
    }
  }
  {
    TopAbs_ShapeEnum shape_type = (&result)->ShapeType();
```
Looks like result is null: 
[image]

Any ideas?

adam-urbanczyk, 2018-12-21:
I am not a SWIG expert, but looked around a bit and found this:https://github.com/tpaviot/pythonocc-core/blob/5416ecc31bacf5349367e20de0220c026189e317/src/SWIG_files/common/FunctionTransformers.i#L130

This could be the rootcause - if TopoDS_Shape in question is null the ` $1.ShapeType()` call will result in segfault. @tpaviot any thougths?

adam-urbanczyk, 2019-01-12:
#605 does remove the crash, but it is not a correct way of handling things. I get the following error now:
```
SystemError: <built-in function HLRBRep_HLRToShape_Rg1LineVCompound> returned NULL without setting an error
```
cf. https://ci.appveyor.com/project/adam-urbanczyk/cadquery-icqj5/builds/21571387#L1264

I assume that the typemap should return `None`  ( i.s.o. a null pointer). PR will follow.

closed 2019-01-13 by tpaviot

tpaviot, 2019-01-13:
@adam-urbanczyk I let you close the issue if you consider it's fixed

adam-urbanczyk, 2019-01-13:
@tpaviot yes, everything looks good now

PR #614 (2019-01-13): Use Py_RETURN_NONE macro in the TopoDS_Shape typemap
This PR should resolve #600