tpaviot/pythonocc-core #602: PrintToString() broken? (for TCollection_ExtendedString)

donalffons, 2018-12-17:
Hi!

Thomas, first of all, thanks a lot for your great project.

I have been trying to reconstruct the product hierarchy of a STEP file. I saw that this question has been raised several times. I am using pythonocc version 0.18.2. I found a (supposedly) good example [here on github](https://github.com/hinxx/pythonocc-core/blob/34abbebb5287bb7f6f808648238e093aa4bd8fe0/examples/core_load_step_ap214_ocaf.py) - although it is a bit dated. The example works well, if I comment out [Line 88](https://github.com/hinxx/pythonocc-core/blob/34abbebb5287bb7f6f808648238e093aa4bd8fe0/examples/core_load_step_ap214_ocaf.py#L88). With that line (specifically, when calling n.Get().PrintToString()), the program crashes.

I have two questions:

1. Does the example still (given its age) show a proper way of extracting product structure and naming from STEP files?
2. What can I do to make it work?

Any help would be greatly appreciated. Cheers
-Sebastian

tpaviot, 2018-12-17:
@donalffons TActually this example is outdated, the releated code has been moved back to pythonocc-core in a function named ```read_step_file_with_names_colors```, see https://github.com/tpaviot/pythonocc-core/blob/master/src/Extend/DataExchange.py#L108

Let me remove this example and replace it with another one up-do-date.

donalffons, 2018-12-17:
Hi @tpaviot, thanks for the links. I used the [core_load_step_ap203_ocaf.py](https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_load_step_ap203_ocaf.py) example to try to extract names of faces and still face the same issue.

I would expect this code to output the names of all root shapes.
[image]

Here is the output:
[image]
...and the execution never completes. Tried it with different STEP files.

[Full Code](https://pastebin.com/hR0v6sMe)

Is that a mistake on my side or a real issue with pythonocc?

donalffons, 2018-12-27:
I did a bit more investigation. The issue seems to be with Handle_TDataStd_Name. Even the following simple code crashes (in the last line):

```python
from OCC.Core.TDataStd import Handle_TDataStd_Name
from OCC.Core.TCollection import TCollection_ExtendedString
N = Handle_TDataStd_Name()
N.Set(TCollection_ExtendedString("Test")) 
```

@tpaviot, could this be an issue with the Swig bindings? I am using pythonocc-core (from -c tpaviot) and oce=0.18.3 (from -c tpaviot). Do you have recommendations what I could try to solve this?

Happy Holidays :)!

donalffons, 2018-12-27:
...also trying pythonocc-core=0.18.2 with oce=0.18.2 does not work.

tpaviot, 2018-12-27:
This one should work:
```
from OCC.Core.TDataStd import TDataStd_Name
from OCC.Core.TCollection import TCollection_ExtendedString
N = TDataStd_Name()
s = TCollection_ExtendedString("Test")
a = N.Get()
a.SetValue(1, s)
print(a.PrintToString())
```

donalffons, 2018-12-27:
My goal is to traverse the assembly hierarchy of a step file and display the names of the individual parts. Here is a view of [OCC CAD assistant](https://www.opencascade.com/content/cad-assistant) for the "as1-oc-214.stp" example
[image]
It shows a total of 9 distinct components in the product hierarchy tree.

I would expect the following code (minimal example) to output the names of these 9 components:
```python

from OCC.Core.TDataStd import TDataStd_Name, TDataStd_Name_GetID
from OCC.Core.TCollection import TCollection_ExtendedString
from OCC.Core.TDocStd import TDocStd_Document
from OCC.Core.XCAFDoc import XCAFDoc_DocumentTool_ShapeTool
from OCC.Core.STEPCAFControl import STEPCAFControl_Reader
from OCC.Core.IFSelect import IFSelect_RetDone
from OCC.Core.TDF import TDF_LabelSequence

filename = 'pythonocc-demos/assets/models/as1-oc-214.stp'

doc = TDocStd_Document(TCollection_ExtendedString("pythonocc-doc"))
shape_tool = XCAFDoc_DocumentTool_ShapeTool(doc.Main())

step_reader = STEPCAFControl_Reader()
step_reader.SetNameMode(True)

status = step_reader.ReadFile(filename)
if status == IFSelect_RetDone:
    step_reader.Transfer(doc)
    print("Step file loaded")
else:
    print("Step file did not load")
    exit()
    
labels = TDF_LabelSequence()
shape_tool.GetShapes(labels)

N = TDataStd_Name()

label1 = labels.Value(1)
print("Number of Shapes: " + str(labels.Length()))
for i in range(1, labels.Length()+1):
    print("Number: " + str(i))
    print("    has name attribute: " + str(labels.Value(i).FindAttribute(TDataStd_Name_GetID(), N)))
    print("Name: " + N.Get().PrintToString())
```

Instead I get:
```
Step file loaded
Number of Shapes: 9
Number: 1
    has name attribute: True
Name: 
Number: 2
    has name attribute: True
Name: 
Number: 3
    has name attribute: True
Name: 
Number: 4
    has name attribute: True
Name: 
Number: 5
    has name attribute: True
Name: 
Number: 6
    has name attribute: True
Name: 
Number: 7
    has name attribute: True
Name: 
Number: 8
    has name attribute: True
Name: 
Number: 9
    has name attribute: True
Name: 
```
All the names are empty...

tpaviot, 2018-12-28:
I understand what you need, but I don't know why names are empty. I also noticed that on my side. However, I don't know any other implementation to get label names.

w-z-l, 2019-08-12:
I have the same question

w-z-l, 2019-08-13:
Excuse me ,this problem weather is solved

tpaviot, 2019-12-29:
The STEP part names issues has been solved by commit https://github.com/tpaviot/pythonocc-core/commit/fb362a5b2e91e9f1550d2364b442ed2c3b7e470c

See @dblanding https://github.com/dblanding/cadviewer project, I guess it's what you need
[image]

closed 2019-12-29 by tpaviot