tpaviot/pythonocc-core #608: one surface with different color

liuxin2322, 2019-01-06:
just like this, how to realize it ?
[image]
i think maybe it can be included in occ examples

tpaviot, 2019-01-06:
You're right, such an example would be welcome. The idea is to to create a ```MeshVS_Mesh``` class, set colors for each note, and display a ```V3d_ColorScale```. You can have a look at http://www.it610.com/article/2029169.htm

Example for the color scale:
```

from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.AIS import AIS_Shape
from OCC.Core.Quantity import Quantity_NOC_BLACK
from OCC.Core.gp import gp_XY

from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu = init_display()

myBox = BRepPrimAPI_MakeBox(60, 60, 50).Shape()

view = display.View
colorscale = view.ColorScale()

aMinRange    = colorscale.GetMin()
aMaxRange    = colorscale.GetMax()
aNbIntervals = colorscale.GetNumberOfIntervals()
aTextHeight  = colorscale.GetTextHeight()
labPosition = colorscale.GetLabelPosition()
position =  gp_XY(colorscale.GetXPosition(), colorscale.GetYPosition())
title = colorscale.GetTitle()

view.ColorScaleDisplay()
display.DisplayShape(myBox)
start_display()
```

liuxin2322, 2019-01-06:
well, very thanks for your answering. using your code ,i obtain color bar smoothly as below.
[image]
well,actually i can't master coloring surface, could you show me more code about painting the surface . it's so difficult for me to realize (and i have read the article you posted 4 days ago and have no ability to convert c++ code to pythonocc. embarassing! :worried: )

sstolzenberg, 2020-11-13:
Hello Thomas,
nice example! I am using PythonOCC 7.4.0, so using
colorscale = view.ColorScale()
throws
"AttributeError: 'V3d_View' object has no attribute 'ColorScale'"
- where has ColorScale object moved?
Thanks for your help,
Best,
Sebastian

tpaviot, 2020-11-14:
It runs on my machine, at least with current master branch. Are you sure you run 7.4.0 ?

sstolzenberg, 2020-11-14:
should be working now, thanks!

closed 2020-11-14 by tpaviot

Tanneguydv, 2023-03-13:
Hello, 
as I can't make the `view.ColorScale()` works, as it seems no longer implemented, and as `MeshDS` is not available for now [#issue1015](https://github.com/tpaviot/pythonocc-core/issues/1015) I've written a small code to color a surface according to a distance criteria, it works better with a `gmsh` mesh but it's a bit longer.

```
from OCC.Core.BRep import BRep_Builder, BRep_Tool
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace, BRepBuilderAPI_MakeWire 
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeSphere, BRepPrimAPI_MakePrism
from OCC.Core.BRepMesh import BRepMesh_IncrementalMesh
from OCC.Core.Graphic3d import Graphic3d_NOM_DEFAULT
from OCC.Core.TopExp import TopExp_Explorer
from OCC.Core.TopAbs import TopAbs_FACE
from OCC.Core.TopoDS import TopoDS_Compound, topods_Face
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Core.gp import gp_Pnt, gp_Vec
from OCC.Core.GeomAPI import GeomAPI_PointsToBSpline
from OCC.Core.TColgp import TColgp_Array1OfPnt

from OCC.Extend.TopologyUtils import TopologyExplorer

from OCCUtils.Common import minimum_distance

from OCC.Display.OCCViewer import rgb_color
from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()

def prism():
    # the bspline profile
    array = TColgp_Array1OfPnt(1, 5)
    array.SetValue(1, gp_Pnt(0, 0, 0))
    array.SetValue(2, gp_Pnt(20, 20, 0))
    array.SetValue(3, gp_Pnt(40, 30, 0))
    array.SetValue(4, gp_Pnt(60, 20, 0))
    array.SetValue(5, gp_Pnt(80, 00, 0))
    bspline = GeomAPI_PointsToBSpline(array, 3, 8).Curve()
    profile = BRepBuilderAPI_MakeEdge(bspline).Edge()

    # the linear path
    starting_point = gp_Pnt(0.0, 0.0, 0.0)
    end_point = gp_Pnt(0.0, 0.0, 60.0)
    vec = gp_Vec(starting_point, end_point)
    path = BRepBuilderAPI_MakeEdge(starting_point, end_point).Edge()

    # extrusion
    prism = BRepPrimAPI_MakePrism(profile, vec).Shape()
    return prism

def simple_mesh(shape):
    # Mesh the shape
    BRepMesh_IncrementalMesh(shape, 0.1)
    builder = BRep_Builder()
    comp = TopoDS_Compound()
    builder.MakeCompound(comp)

    bt = BRep_Tool()
    ex = TopExp_Explorer(shape, TopAbs_FACE)
    while ex.More():
        face = topods_Face(ex.Current())
        location = TopLoc_Location()
        facing = bt.Triangulation(face, location)
        tri = facing.Triangles()
        for i in range(1, facing.NbTriangles() + 1):
            wirebuild = BRepBuilderAPI_MakeWire()
            trian = tri.Value(i)
            index1, index2, index3 = trian.Get()
            for j in range(1, 4):
                if j == 1:
                    m = index1
                    n = index2
                elif j == 2:
                    n = index3
                elif j == 3:
                    m = index2
                me = BRepBuilderAPI_MakeEdge(facing.Node(m), facing.Node(n))
                if me.IsDone():
                    wirebuild.Add(me.Edge())
            faace = BRepBuilderAPI_MakeFace(wirebuild.Shape()).Shape()
            builder.Add(comp, faace)
        ex.Next()
    return comp

def get_rgb_color_from_scale(value, bounds=[0, 100], colors=[(26, 54, 77), (77, 37, 26)]):
    for i in range(len(bounds)-1):
        if bounds[i] <= value < bounds[i+1]:
            ratio = (value - bounds[i]) / (bounds[i+1] - bounds[i])
            color = [int((colors[i+1][j] - colors[i][j]) * ratio + colors[i][j]) for j in range(3)]
            rgb = [i/255 for i in color]
            return tuple(rgb)
    color = colors[-1]
    rgb = [i/255 for i in color]
    return tuple(rgb)

def colored_mesh_distance(comp, ref):
    distances = []
    for fc in TopologyExplorer(comp).faces():
        dist, _, _ = minimum_distance(pt_sphere, fc)
        distances.append(dist)     
    # create bounds for the color scale
    bounds = [min(distances), max(distances)]
    display.default_drawer.SetFaceBoundaryDraw(False)
    for fc, dist in zip(TopologyExplorer(comp).faces(), distances):
        color = get_rgb_color_from_scale(dist, bounds=bounds)
        display.DisplayShape(fc, color=rgb_color(*color), material=Graphic3d_NOM_DEFAULT)
    display.FitAll()

if __name__ == "__main__":
    #Create shape
    shape = prism()
    #Create a compound of faces from the shape meshed
    comp = simple_mesh(shape)
    # Create reference shape
    point = gp_Pnt(40, 15, 30)
    pt_sphere = BRepPrimAPI_MakeSphere(point, 0.5).Shape()
    display.DisplayShape(pt_sphere)
    #color a face according to its distance to the ref
    colored_mesh_distance(comp, pt_sphere)
    start_display()
```
the result : 
[image]

if interested I could push to the demos